wokwi/wokwi-features #653: std::array won't work

AlexanderTonn, 2023-10-09:
It's not possible to include the std::array header in the wokwi online editor.  With Arduino IDE 2.0 and VSCode Arduino extension, the ```#include <array>``` is compiling without error

I've created a project for showing the issue:  https://wokwi.com/projects/377839112927963137

``` shell
sketch.ino:1:10: fatal error: array: No such file or directory
 #include <array>
          ^~~~~~~
compilation terminated.

Error during build: exit status 1
```

urish, 2023-10-09:
`<array>` is also not available in the Arduino IDE when compiling for the Arduino Board Uno:

[image]

You may have installed some third party library that provides this file into your local Arduino installation.

Behind the scenes, Wokwi uses the Arduino CLI to compile your code, so it should be equivalent to how Arduino IDE builds the code. You can use the [Library Manager](https://docs.wokwi.com/guides/libraries) to install standard Arduino libraries as needed.

sutaburosu, 2023-10-09:
I have used the code [in this Stackexchange comment](https://arduino.stackexchange.com/a/69178) successfully on Wokwi.