wokwi/wokwi-features #656: stubborn wire in diagram

alto777, 2023-10-09:
This project

[This project](https://wokwi.com/projects/378031652666068993)

has a wire I have tried several times to fix. The wire from resistor dDR1 to Vdd.

Moving it does not light up the "save" button - I though I was missing having saved my work. Making another trivial change to get the "save" button to be active was my way around that.

But the wire was back later. I believe the diagram.json file is not being changed when this wire is moved.

I hope it is not so elusive that one cannot open the project, move the wire and save, only to see it come back. It's the knida thing where deleting the wire and placing a new one might fix it.

MacOS 10.14.5 Chrome Version 116.0.5845.187 (Official Build) (x86_64)

I report this trivial nuisance since it may be only the tip of an iceberg.

alto777

urish, 2023-10-09:
Issue reproduced here. You seem to hit some edge case where the wire movement is not really terminated properly (thus, Undo doesn't work as well). looking into this.

urish, 2023-10-10:
Found the culprit: the following line should generate a memory access exception, but it doesn't (at least in RISC-V)

https://github.com/espressif/esp-idf/blob/1ec70cd5681c931560b0a1c758de2277b332ccd5/components/esp_system/panic.c#L452

alto777, 2024-12-20:
I found this searching for "wokwi wire can't be moved". I had forgotten all about the issue. Now I have a project where no wires can be moved, not exist wires or newly placed wires.

[Immovable wires project](https://wokwi.com/projects/417805966639647745)

alto777

urish, 2024-12-21:
> Now I have a project where no wires can be moved, not exist wires or newly placed wires.

That was a regression we had across Wokwi, should be fixed now.

alto777, 2024-12-21:
The wires move in the linked project now.

alto777

urish, 2025-01-13:
Minimal reproduction: https://wokwi.com/projects/419992016430556161