wokwi/wokwi-features #657: ESP Goes to Sleep after WiFi Connection

apollolabsdev, 2023-10-09:
In an HTTP client example on Wokwi, after configuring and connecting to WiFi using station mode, the remainder code is not executed. Instead, the ESP32 goes to sleep.

[image]

Viewing the packet trace on Wokwi, the last packet sent by the ESP is an ARP announcement before it goes to sleep.

[image]

Link to Simulation Project: https://wokwi.com/projects/377728497260065793

I tested the same code on hardware and the remainder code is executed as expected. An http client is set up and webpage data is retrieved. The device's sleep behavior is not replicated.

Device is ESP32C3
Language is Rust
Browser is Chrome

urish, 2023-10-10:
Quick check with GDB shows that this crashes due to stack overflow:

```
Remote debugging using localhost:3333
__ubsan_include ()
    at /data/espressif/esp-idf/v4.4.6/components/esp_system/ubsan.c:294
294     /data/espressif/esp-idf/v4.4.6/components/esp_system/ubsan.c: No such file or directory.
(gdb) bt
#0  __ubsan_include ()
    at /data/espressif/esp-idf/v4.4.6/components/esp_system/ubsan.c:294
#1  0x4038601e in esp_system_abort (
    details=details@entry=0x3fc924c4 <xIsrStack+1380> "***ERROR*** A stack overflow in task sys_evt has been detected.")
    at /data/espressif/esp-idf/v4.4.6/components/esp_system/esp_system.c:137
#2  0x40388bec in vApplicationStackOverflowHook (
    xTask=<optimized out>, pcTaskName=<optimized out>)
    at /data/espressif/esp-idf/v4.4.6/components/freertos/port/riscv/port.c:357
#3  0x40387b98 in vTaskSwitchContext ()
    at /data/espressif/esp-idf/v4.4.6/components/hal/esp32c3/include/hal/cpu_ll.h:36
#4  0x4038609a in isr_skip_decrement ()
    at /data/espressif/esp-idf/v4.4.6/components/freertos/port/riscv/portasm.S:95
```

I'm going to test the same binary on the actual chip and compare

urish, 2023-10-10:
Tested on a physical chip (with the [real WOKWI-Guest](https://twitter.com/WokwiMakes/status/1687061850951720960)), and it also crashes. Full log: 

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038190e
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5820,len:0x171c
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f68
SHA-256 comparison failed:
Calculated: 1d06b938c0222bf626e0bdf46178b1b37ab24d03f0360fc8fcf7153c2571deaf
Expected: 68d7bdf643ba446b8ed7ae8423241d442fd052b2bc77091100ba06fd65dcf8d5
Attempting to boot anyway...
entry 0x403cc710
I (53) boot: ESP-IDF v5.1-beta1-378-gea5e0ff298-dirt 2nd stage bootloader
I (53) boot: compile time Jun  7 2023 07:59:10
I (55) boot: chip revision: v0.3
I (59) boot.esp32c3: SPI Speed      : 40MHz
I (63) boot.esp32c3: SPI Mode       : DIO
I (68) boot.esp32c3: SPI Flash Size : 4MB
I (73) boot: Enabling RNG early entropy source...
I (78) boot: Partition Table:
I (82) boot: ## Label            Usage          Type ST Offset   Length
I (89) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (97) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (104) boot:  2 factory          factory app      00 00 00010000 003f0000
I (112) boot: End of partition table
I (116) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=35ba0h (220064)                                                                                                              map
I (173) esp_image: segment 1: paddr=00045bc8 vaddr=3fc8ec00 size=02438h (  9272)                                                                                                              load
I (175) esp_image: segment 2: paddr=00048008 vaddr=40380000 size=08010h ( 32784)                                                                                                              load
I (186) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=c2bc8h (797640)                                                                                                              map
I (361) esp_image: segment 4: paddr=00112bf0 vaddr=40388010 size=06b48h ( 27464)                                                                                                              load
I (373) boot: Loaded app from partition at offset 0x10000
I (373) boot: Disabling RNG early entropy source...
I (384) cpu_start: Unicore app
I (385) cpu_start: Pro cpu up.
I (393) cpu_start: Pro cpu start user code
I (393) cpu_start: cpu freq: 160000000
I (394) cpu_start: Application information:
I (396) cpu_start: Project name:     libespidf
I (401) cpu_start: App version:      1
I (406) cpu_start: Compile time:     Oct 10 2023 11:58:24
I (412) cpu_start: ELF file SHA256:  0000000000000000...
I (418) cpu_start: ESP-IDF:          v4.4.6-dirty
I (423) cpu_start: Min chip rev:     v0.3
I (428) cpu_start: Max chip rev:     v0.99
I (433) cpu_start: Chip rev:         v0.3
I (438) heap_init: Initializing. RAM available for dynamic allocation:
I (445) heap_init: At 3FC95860 len 00046EB0 (283 KiB): DRAM
I (451) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (458) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (465) spi_flash: detected chip: generic
I (469) spi_flash: flash io: dio
I (473) sleep: Configure to isolate all GPIO pins in sleep state
I (479) sleep: Enable automatic switching of GPIO sleep configuration
I (486) cpu_start: Starting scheduler.
I (498) pp: pp rom version: 9387209
I (498) net80211: net80211 rom version: 9387209
I (508) wifi:wifi driver task: 3fc9dd40, prio:23, stack:6656, core=0
I (508) system_api: Base MAC address is not set
I (518) system_api: read default base MAC address from EFUSE
I (528) wifi:wifi firmware version: 1ba8b6a
I (528) wifi:wifi certification version: v7.0
I (528) wifi:config NVS flash: enabled
I (528) wifi:config nano formating: disabled
I (538) wifi:Init data frame dynamic rx buffer num: 32
I (538) wifi:Init management frame dynamic rx buffer num: 32
[... 58 lines cut ...]
3fc92660: 0x3fc92658 0x00000000 0x3fc9266c 0xffffffff 0x3fc9266c 0x3fc9266c 0x00                                                                                                             000000 0x3fc92680
3fc92680: 0xffffffff 0x3fc92680 0x3fc92680 0x00000000 0x3fc92694 0xffffffff 0x3f                                                                                                             c92694 0x3fc92694
3fc926a0: 0x00000000 0x3fc926a8 0xffffffff 0x3fc926a8 0x3fc926a8 0x00000000 0x3f                                                                                                             c926bc 0xffffffff
3fc926c0: 0x3fc926bc 0x3fc926bc 0x00000001 0x3fc926d0 0xffffffff 0x3fca42c4 0x3f                                                                                                             ca42c4 0x00000000
3fc926e0: 0x3fc926e4 0xffffffff 0x3fc926e4 0x3fc926e4 0x00000000 0x3fc926f8 0xff                                                                                                             ffffff 0x3fc926f8
3fc92700: 0x3fc926f8 0x00000000 0x3fc9270c 0xffffffff 0x3fc9270c 0x3fc9270c 0x00                                                                                                             000000 0x3fc92720
3fc92720: 0xffffffff 0x3fc92720 0x3fc92720 0x00000000 0x3fc92734 0xffffffff 0x3f                                                                                                             c92734 0x3fc92734
[... cut ...]
3fc92860: 0x00000009 0x3fc96228 0x3fc96290 0x3fc962f8 0x00000000 0x00000000 0x00                                                                                                             000001 0x00000000
3fc92880: 0x00000000 0x00000000 0x420b441e 0x00000000 0x00000000 0x00000000 0x00                                                                                                             000000 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
```

So it doesn't seem to be a simulator bug. I'm keeping it open though - as I want to investigate why the crash info is not printed in the simulator.

urish, 2023-10-10:
Found the culprit: the following line should generate a memory access exception, but it doesn't (at least in RISC-V)

https://github.com/espressif/esp-idf/blob/1ec70cd5681c931560b0a1c758de2277b332ccd5/components/esp_system/panic.c#L452

urish, 2023-10-10:
Fix is up - now the simulator prints the exception information just like the real hardware

closed 2023-10-10 by urish