wokwi/wokwi-features #662: Issue with ILI9341 display

AlphaTechNinja, 2023-10-14:
**Describe the bug**
When I attempt to connect a ILI9341 to Raspberry Pi Pico (not in MicroPython) and use the Adafruit GFX library it won’t draw anything when I any function in the library

**To Reproduce**
Make a project using the Raspberry Pi Pico (not MicroPython) and wire up a ILI9341 display to the arduino SPI pins (not the ones the MCU uses for serial communication which are the first few pins) and create a new Adafruit_ILI9341 object use every necessary pin that includes DC,CS,MOSI,MISO,SCK and RST in order DC,CS,MOSI,SCK,RST,MISO (I might have MOSI and MISO swapped) and use the “begin” method in setup function and try using a GFX function

**Expected behavior**
I should draw or print something to the screen
Like it does in real life

**Environment (please complete the following information):**
 - OS: iOS
 - Browser:safari
 - Version:latest

leftCoast, 2023-10-14:
This uses the Adafruit GFX stuff.
https://wokwi.com/projects/362198011666176001

urish, 2023-10-14:
@AlphaTechNinja please share a link to a project that demonstrates the issue.

AlphaTechNinja, 2023-10-14:
Here I try making a quick example of it I have verified that pins are correct and try drawing things to it https://wokwi.com/projects/378575342731426817

urish, 2023-10-14:
Thanks for making an example. DC and CS are swapped. If you change lines 6-7 to read:

```cpp
#define TFT_DC 20
#define TFT_CS 22
```

Then it works correctly:

[image]

closed 2023-10-19 by urish