wokwi/wokwi-features #664: TFT_eSPI library with ESP32 and hardware SPI causes garbled screen.

Koepel, 2023-10-19:
The newest TFT_eSPI library on a ESP32 with hardware SPI causes a garbled screen in Wokwi.
The bug was mentioned in the Discord channel for a ESP32-S3, but a standard ESP32 has the same problem.
It think that a older version of the library did work.

urish, 2023-10-19:
Reproduction: https://wokwi.com/projects/373508684566814721

urish, 2023-10-19:
From initial investigation, it doesn't look like a regression, at least not a recent one.

Koepel, 2023-10-19:
Someone uploaded a TFT_eSPI library which works: https://wokwi.com/projects/357946487875751937

urish, 2023-10-26:
Would be interesting to compare that library (can be downloaded through the Library Manager tab) with the faulty one