wokwi/wokwi-features #669: Raspberry PI micropython error

AgentXI, 2023-11-02:
**Describe the bug**
I am using a Raspberry Pi with Micropython. When I import the time module and I print out time.time(), the value of time.time() is always 1609459200.

**To Reproduce**
1: Open a new project with a Raspberry PI and Micropython.
2: Insert this code in main.py
"""
import time
time.sleep(0.1)

while True:
    print(time.time())
    time.sleep(0.1)
"""
3: Run the code
Example Project: https://wokwi.com/projects/380069958541541377

**Expected behavior**
I thought the code would print different values in the console like, 1745459235.

**Environment (please complete the following information):**
 - OS: ChromeOS on a Chromebook
 - Browser: Chrome
 - Version: Version 118.0.5993.86

urish, 2023-11-03:
Probably related to the fact we don't implement (yet) the RTC peripheral

urish, 2023-11-11:
Fixed!

closed 2023-11-11 by urish