wokwi/wokwi-features #677: Simulator not recignuising variable types

Graham55H, 2023-11-14:
The simulator changes global variables in an interrupt routine which are then reflected in the body of the program. In the actual device, changes made in an interrupt routine are NOT reflected in the body of the program unless the variable is declared as "volatile".

I delcare a variable "milliseconds" as a global unsigned long, and set it to 0. In the interrupt routine, this variable is incremented on a timer interrupt when a switch is closed. In the main program, this initiates the "loop" action. In the simulator, closing the switch sets the loop process in action, but in the actual device, the loop doesn't run. I traced this to the change in the "milliseconds" variable not being transferred out of the interupt; declaring the "milliseconds" variable as "volatile" corrects the problem, and allows the loop code to run.

Windows 10, Chrome version 119, Arduino Uno, rev 3. The device compiler / programmer is the Arduino IDE, V2.2.1. The code can be found under MIDI_player in my projects.

sutaburosu, 2023-11-17:
> changes made in an interrupt routine are NOT reflected in the body of the program unless the variable is declared as "volatile".

This is to be expected, and exactly what the `volatile` keyword is intended to work around. Honestly, it's surprising that your code works as intended on your physical device without `volatile`.

I feel that this is unlikely to be an issue with the simulation, but rather a difference in the compiled binaries generated by your local Arduino toolchain compared to the toolchain installed on the Wokwi build server.

It would be nice if you could verify this by uploading [the .hex built on your machine](https://arduino.stackexchange.com/questions/48431/how-to-get-the-firmware-hex-file-from-a-ino-file-containing-the-code#48437) and testing it within Wokwi. Press F1 in the Wokwi code editor and select "Upload Firmware and Start Simulation".

> The code can be found under MIDI_player in my projects.

I checked your GitHub profile, and there are no repos/projects with that name, so I presume you're referring to a Wokwi project. Your username here on GitHub isn't listed on [the Wokwi Makers page](https://wokwi.com/makers) so there is no way for me to discover your projects. It would be useful to include a link to your project in this ticket.

Graham55H, 2023-11-17:
Hi – see my inline response in red.

 

Best regards

 

Graham

 

From: sutaburosu ***@***.***> 
Sent: 17 November 2023 19:24
To: wokwi/wokwi-features ***@***.***>
Cc: Graham55H ***@***.***>; Author ***@***.***>
Subject: Re: [wokwi/wokwi-features] Simulator not recognising variable attributes correctly (Issue #677)

 

changes made in an interrupt routine are NOT reflected in the body of the program unless the variable is declared as "volatile".

This is to be expected, and exactly what the volatile keyword is intended to work around. Honestly, it's surprising that your code works as intended on your physical device without volatile.

It doesn’t – that was my point.  I’m a beginner with C, and with the Arduino series of microcomputers, and I expected the simulator to be a reliable way of debugging my code and helping me learn.  The simulator worked as intended both with and without the volatile keyword, but the actual Arduino device only worked when it was included (once I discovered it was necessary!!) ... it was therefore giving misleading information.

I feel that this is unlikely to be an issue with the simulation, but rather a difference in the compiled binaries generated by your local Arduino toolchain compared to the toolchain installed on the Wokwi build server.

I’m sure that’s right ... but the simulator was still giving misleading information.  If global variables need to be declared as volatile to make then usable in an interrupt routine – which I believe is the function of the keyword – that ought to be reflected in the operation of any simulator.  The source of the binaries surely shouldn’t be relevant?  Otherwise, what’s the point of having a language specification?

It would be nice if you could verify this by uploading the .hex built on your machine <https://arduino.stackexchange.com/questions/48431/how-to-get-the-firmware-hex-file-from-a-ino-file-containing-the-code#48437>  and testing it within Wokwi. Press F1 in the Wokwi code editor and select "Upload Firmware and Start Simulation".

Thank you for the suggestion – I’ll do that.

sutaburosu, 2023-11-18:
> The source of the binaries surely shouldn’t be relevant?  Otherwise, what’s the point of having a language specification?

It is relevant. Different minor versions of a single compiler can generate very different code from the same source. There is nothing in the specification preventing a compiler from reading a variable from memory each time it is referenced (i.e. code that unexpectedly works without `volatile`). For some reason the compiler chose to not optimise away these superfluous memory accesses, despite not being forced to do so by `volatile`.

I'm sure if you upload the hex that didn't work on a real board, you will see the same results in Wokwi.

closed 2023-11-24 by urish