wokwi/wokwi-features #681: Dht22 error constants

Gavin-Perry, 2023-11-26:
**Describe the bug**
In dht22.h the error constants e.g. dht22.OK were made private

**To Reproduce**
  if (dht22.getLastError() != dht22.OK) {   // dht22.OK was made private, get compile error

**Expected behavior**
it should compile!

**Environment (please complete the following information):**
Not that it matters...
 - OS: Windows
 - Browser chrome  119.0.6045.160 (Official Build) (64-bit) (cohort: Stable)

urish, 2023-11-26:
Can you get the code to compile on your own computer?

Wokwi doesn't implement its own compiler - it just uses the standard Arduino compiler (arduino-cli).

Gavin-Perry, 2023-11-28:
Yes the code compiles fine on my own computer. Or at least it did. I'll double check to make sure they're haven't been any changes. Perhaps you use different flags in the compiling commands.

Gavin

Sent via the Samsung Galaxy S22 5G, an AT&T 5G smartphone

Phetdanai, 2023-12-29:
Yes but in function `void loop` has this code

`if (isnan(t) || isnan(h)) {`                      
`   Serial.println("Failed to read from DHT sensor");`
`   return;`
`}`

It's my  ESP32 Blynk IoT project

Phetdanai, 2023-12-29:
Serial monitor display
`Failed to read from DHT sensor`
forever

Phetdanai, 2024-01-01:
Only one solution is:
Use pull-up resistor about 4700 - 10000 ohms

Gavin-Perry, 2024-01-01:
Non of the comments addressed the issue of failing to compile. I reinstalled Arduino and rebooted and that resolved the problem.

closed 2024-01-01 by Gavin-Perry