wokwi/wokwi-features #691: Error installing Library - not found

GyverLibs, 2023-12-10:
**Describe the bug**
Error installing library

**To Reproduce**
Bug is hard to reproduce, it can appear after adding an empty line of code in editor, and disappear same way

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: Win 11 x64
 - Browser Chrome
 - Version latest

**Additional context**
[image]

urish, 2023-12-10:
Thanks for reporting! Is this a library that was recently published?

GyverLibs, 2023-12-10:
Few weeks ago. It works in wokwi, but very unstable. Even an empty line of code can break it. And fix too 😅

urish, 2023-12-10:
Thanks, looking into it!

GyverLibs, 2023-12-10:
🙏 🙏

urish, 2023-12-10:
Ok, I uploaded a fix. Can you please try now?

Note that for existing projects, the error might be cached. In that case, simply add `?build-cache=disable`  (e.g. https://wokwi.com/projects/328451800839488084?build-cache=disable) to force a rebuild. This all also update the cache and fix the error in the future.

GyverLibs, 2023-12-10:
yes, it works! Many thanks

closed 2023-12-10 by urish