wokwi/wokwi-features #693: Wokwi Automated Testing with Github Actions

forntoh, 2023-12-13:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when a PR is submitted and all my unit tests pass but still end up breaking my library because not everything can be caught through unit tests alone.

**Describe the solution you'd like**
I would love to have a way to write a script that runs actions in Wokwi autonomously, it will help library developers to ensure that their libraries always work as intended and are not broken in a PR.
For example, say I am developing a menu library, I would love to provide a wokwi schema that has a keypad, display, and controller and a script that runs say an example project file using my library and test if the result displayed on the LCD screen is desired. This can be run on Git Hub whenever a PR is created and can block or allow a PR depending on the result.

urish, 2023-12-17:
Hi, have you looked at [Wokwi CI](https://docs.wokwi.com/wokwi-ci/getting-started)?

forntoh, 2023-12-21:
I can't figure out how to use the examples as test files, in my library I have multiple ino examples which showcase how to use the library, I want to test all the examples with scenarios, but I don't know how to pass the path of each example file to be used for each scenario I want to run. In my project, the src directory only contains the necessary code for the library (it's not executable).

forntoh, 2023-12-21:
I also don't see any documentation about the options available for the scenario file
I can see `wait-serial`, but I am not sure if `set-serial` exists, I want to control the microcontroller using serial input.

closed 2025-11-03 by urish