wokwi/wokwi-features #696: Wokwi Traditional Chinese Translation

Sear-0887, 2023-12-17:
**Is your feature request related to a problem? Please describe.**
Please give the community using non-simplified chinese a translation.
Note the #211 #393 all discusses about simplified chinese which is different.
I've translated the current English string json and some preview:
[zhcn.json](https://github.com/wokwi/wokwi-features/files/13696702/zhcn.json)
```
...
"project.shareDialog.closeButton": "關閉",
  "simulator.tabs.simulation": "模擬",
  "simulator.tabs.code": "程式",
  "simulator.tabs.description": "描述",
  "simulator.main.restart": "重啟模擬",
  "simulator.main.start": "開始模擬",
  "simulator.main.resume": "繼續",
  "simulator.main.pause": "暫停",
  "simulator.main.addNewPart": "增加組件",
  "simulator.main.stopSimulation": "停止模擬",
  "simulator.main.zoomGridTooltip": "縮放, 網格與更多",
  "simulator.buildFailedDialog.title": "建構失敗",
...
```

urish, 2023-12-26:
Thanks!

Translations added :)

[image]

closed 2023-12-26 by urish

urish, 2023-12-26:
Direct link to see the translation in action: https://wokwi.com/projects/new/esp32?lang=zh-Hant