wokwi/wokwi-features #698: Wokwi for VS Code: virtual serial port terminal since version 2.1.0 does not show input (serial monitor)

dhebbeker, 2023-12-19:
### Describe the bug

The [Wokwi for VS Code extension](https://marketplace.visualstudio.com/items?itemName=wokwi.wokwi-vscode) uses VS Code's build-in terminal since the version 2.1.0.

With that terminal users can not see what they are writing to the serial port. Writing works: One can write "blindly", press enter and it is sent. But it is cumbersome to write s.th. blindly. The previous version had an extra field where one could write a message before it was sent to the serial port.

#### Screenshot of the old version

[image]

Note the field with orange border and white background where one could enter a message.

#### Screenshot of new version

[image]

Note that such an input line is missing. If the cursor is set into that terminal panel, one can write. But nothing is shown.

### To Reproduce

1. Install the VS Code extension version 2.1.0 or newer
2. Start any Wokwi project with the VS Code extension which accepts input via serial port.
3. Try to type some message to the serial port. It is not displayed. 
4. Press ENTER or RETURN and the message will be sent.

### Expected behavior

One can see the typed message, and be able to modify it before sending.

### Environment

 - OS: Ubuntu and Windows 11
 - VS Code 1.85.1

urish, 2023-12-20:
Thanks for the detailed report! 

We can probably solve this by enabling local echo in the terminal. The issue is that this is non standard, and this will break firmwares that do not expect local echo (such as micropython, Zephyr Project, etc.)

dhebbeker, 2023-12-20:
Would this allow the message to be visible and mutable before sending it, as it was with the previous version?

urish, 2023-12-20:
Not really - the previous version was using a non standard way to handle the serial communication (that was based on the way Arduino IDE does it, which is also non-standard), hence it is not one-to-one compatible with VS Code's built-in terminal.

dhebbeker, 2023-12-21:
I did try to use the [suggested](https://github.com/wokwi/wokwi-features/issues/699#issuecomment-1865835040) control sequence "SRM" `ESC[12l` respectively `\x1b[12l` to enable local echo in the terminal. It seems to have **no effect** to VS Code's built-in terminal. 😔

The ANSI sequence is defined by ANSI X3.64-1979, see [Summary of ANSI standards for ASCII terminals](https://www.inwap.com/pdp10/ansicode.txt).

urish, 2023-12-21:
Wow, 1984. Soon to turn 40!

dhebbeker, 2023-12-21:
For the moment I am using version 2.0.1 of the extension because the input line is essential for my testing. I hope it will stay compatible with Wokwi's server-side services (until this issue is resolved?). 🤞

Can one attach a different terminal to the simulator? For example the [Serial Monitor extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor) for VS Code? This one is customizable and has a dedicated input line.

urish, 2023-12-21:
> For the moment I am using version 2.0.1 of the extension because the input line is essential for my testing. I hope it will stay compatible with Wokwi's server-side services (until this issue is resolved?). 🤞

Yes, should stay compatible at least for the next few months.

> Can one attach a different terminal to the simulator? For example the [Serial Monitor extension](https://marketplace.visualstudio.com/items?itemName=ms-vscode.vscode-serial-monitor) for VS Code? This one is customizable and has a dedicated input line.

Good question - Wokwi can expose the serial port over [RFC2217 Socket](https://docs.wokwi.com/vscode/project-config#serial-port-forwarding), so if the Serial Monitor Extension supports connecting to serial over TCP it might work.

dhebbeker, 2023-12-21:
Great, the Serial Monitor extension does support connecting over TCP! ✅ Thank you for the pointer to the respective Wokwi documentation. 🙏

[image]

I just have a little "problem" with the TCP monitor: When sending a message it is not terminated with `\n` or `\r` or `0` which I used until now to detect the end of the message. This is not directly part of this issue, but maybe you have an idea?

urish, 2023-12-21:
Thanks for trying it out!

Apparently, I already had the extension installed :)  I'll link to it from the docs.

Have you tried the following setting?

[image]

urish, 2023-12-21:
Hmm, apparently not yet for TCP sockets: microsoft/vscode-serial-monitor#130

dhebbeker, 2023-12-21:
Thank you for looking it up. It is a pity it does not work with the Serial Monitor and TCP.

I tested dhebbeker/echo-device@575574aefd3d74b31a029c29450dca1b08cd50c4 with PuTTY SSH Client in Telnet mode and it works. ✅

[image]

But PuTTY is not is not so well integrated with VS Code. Also its window closes when I start the simulator. But I can work with that. Although I preferred Wokwi's serial monitor prior to v2.1.0 😊 But maybe there is even a better VS Code extension. I may search for one on demand. For now I use Wokwi's extension v2.0.1 😉

closed 2025-01-13 by urish