wokwi/wokwi-features #699: Wokwi for VS Code: virtual serial port terminal since version 2.1.0 requires carriage return (CR) for new lines

dhebbeker, 2023-12-19:
### Describe the bug

The [Wokwi for VS Code extension](https://marketplace.visualstudio.com/items?itemName=wokwi.wokwi-vscode) uses VS Code's build-in terminal since the version 2.1.0.

That terminal now expects a carriage return for new lines. Without the new line will continue at the same column where the previous ended.

I am using `std::endl` to terminate lines I send via serial port. This does not work properly any more as `\n\r` is required.

#### Screenshot of the old version

[image]

Note how new lines begin left aligned.

#### Screenshot of new version

[image]

Note how some of the lines start where the previous line left of, but one line further down.

### To Reproduce

1. Install the VS Code extension version 2.1.0 or newer
2. Start any Wokwi project with the VS Code extension which writes lines via serial port ending with `\n`.

### Expected behavior

`\n` starts a new line at the first column.

Ideally one can select the expected line ending.

### Environment

 - OS: Ubuntu and Windows 11
 - VS Code 1.85.1

urish, 2023-12-20:
Thanks for the detailed report!

Can you try printing the following string at the beginning of your program and see if it helps?

`"\x1b[20h"`

This escape code should enable automatic newline conversion.

dhebbeker, 2023-12-21:
✅ Using the string you suggested helps, thank you! 🙏

What is this escape code? Does it have a name? Popular results when searching the web were Wokwi projects using that sequence. 😆

dhebbeker, 2023-12-21:
The behaviour of the serial terminal Wokwi provides or uses has changed with version 2.1.0.

But on second thought I am not sure whether this is actually a bug. May also have been that the previous behaviour was not "correct" but accidentally fitting my needs. I am not even sure if there is a "correct" behaviour in this regard?

I am closing this issues as _I_ do not have sufficient grounds to claim it is a bug.

closed 2023-12-21 by dhebbeker