wokwi/wokwi-features #700: Add a Description/README.md file to the other new project templates

drf5n, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
When I'd doing an Uno project, I have trouble remembering the README.md -> Description tab that you see on :

https://wokwi.com/projects/new/blank

**Describe the solution you'd like**
Please add the README.md file to the other template projects besides "Blank" available from the New Project dropdown on the https://wokwi.com/dashboard/projects page to provide for easy additional documentation.

urish, 2023-12-20:
Thanks for the proposal! What should the README contain?

drf5n, 2023-12-20:
> [EXTERNAL to VIMS received message]
>
> Thanks for the proposal! What should the initial README contain?

Copying a slightly more generalized version of the one that is already delivered in https://wokwi.com/projects/new/blank seems OK:

```
# Wokwi project

This is a Wokwi project. Please edit this README file and add a description of your project.

## Usage

1. Add parts by clicking the blue "+" button in the diagram editor
2. Connect parts by dragging wires between them
3. Click the green play button to start the simulation
```

urish, 2024-01-03:
Thanks! The comment probably doesn't render markdown since it was sent via email:

[image]

Added a default README file for new projects. Let's wait for a week and see how it goes!

e.g. https://wokwi.com/projects/new/esp32

urish, 2024-02-04:
Unfortunately, this change didn't make any significant difference, so I plan to revert it. 

Here's the data:

[image]

closed 2024-02-04 by urish