wokwi/wokwi-features #702: Autosave feature

edemit, 2023-12-22:
It stresses me out that wokwi doesn't implement the autosave feature, which causes me to constantly lose my progress due to external causes. 

Even saving every minute or at most identical saving settings as in vscode would be enough

RaNoL1N, 2023-12-22:
hellp me , i am in his basement and he using me as slave coder.

urish, 2024-01-03:
Thanks for the request!

Are you talking about Wokwi on the Web or Wokwi for VS Code?

edemit, 2024-01-04:
Right now I use Wokwi in web

urish, 2024-01-07:
Got it - we'll need to add some support for versioning first, as you definitely don't want to make some experimental changes to a project, and then discover they were auto saved and the original project no longer exists