wokwi/wokwi-features #706: Add Support for Attribute Dependency in Custom Chips

hayschan, 2023-12-28:
**Describe the feature request:**
I am currently developing a custom chip simulation for an air flow sensor in the Wokwi Simulator. The sensor's behavior involves variable flow rate values depending on another attribute, specifically the 'K' value. The Wokwi Simulator [custom chips AI](https://github.com/wokwi/wokwi-features/issues/301) currently doesn't support attribute dependency, which is crucial for accurately simulating the behavior of such sensors.

**Use Case:**
The air flow sensor has different flow rate ranges based on the 'K' value:

- For K = 1000, the flow rate range is 10SLM to <50SLM.
- For K = 100, the flow rate range is 50SLM to ≤300SLM.

I need to simulate the sensor such that when the 'K' value attribute changes, the range of another attribute (the flow rate) adjusts accordingly. 

Both of these attributes are set by user.

**Current Limitation:**
The current API does not support dynamic adjustment of one attribute's range based on the value of another attribute. Here is the current JSON configuration for the custom chip:

```json
{
    "name": "airFlowSensor",
    "author": "",
    "pins": [
        "GND",
        "SCL",
        "SDA",
        "VCC"
    ],
    "controls": [
        {
            "id": "airFlowRate",
            "label": "Gas Air Flow Rate",
            "type": "range",
            "min": 10,
            "max": 50,
            "step": 1
        },
        {
            "id": "K value",
            "label": "K value",
            "type": "range",
            "min": 100,
            "max": 1000,
            "step": 900
        }
    ]
}
```

**Requested Feature:**
I propose the implementation of attribute dependency within the custom chip configuration. This feature would allow for dynamic changes in one attribute's properties (such as range) based on the value of another attribute. This enhancement would significantly increase the accuracy and utility of simulations involving sensors or components with interdependent parameters.

Thank you for considering this feature request.

djedu28, 2024-10-01:
For this specific application, you can use percentage `airFlowRate`. in the ranger from 0 to 100% for “airFlowRate”.
```json
{
            "id": "airFlowRate",
            "label": "Gas Air Flow Rate",
            "type": "range",
            "min": 10,
            "max": 50,
            "step": 1
 },

```

In this way, the chip's programming will interpret the maximum and minimum `airFlowRate` as varying according to the value of K.

As a visual solution to the `airFlowRate` value, you could use the graphic resource on the chip itself, implemented in [Framebuffer Chip](https://wokwi.com/projects/330503863007183442) - https://wokwi.com/projects/330503863007183442
```json
"display": {
    "width": 128,
    "height": 50
}
```
[image]

>> Translated with DeepL.com

hayschan, 2024-10-15:
> For this specific application, you can use percentage `airFlowRate`. in the ranger from 0 to 100% for “airFlowRate”.
> 
> ```json
> {
>             "id": "airFlowRate",
>             "label": "Gas Air Flow Rate",
>             "type": "range",
>             "min": 0,
>             "max": 100,
>             "step": 1
>  },
> ```
> 
> In this way, the chip's programming will interpret the maximum and minimum `airFlowRate` as varying according to the value of K.
> 
> As a visual solution to the `airFlowRate` value, you could use the graphic resource on the chip itself, implemented in [Framebuffer Chip](https://wokwi.com/projects/330503863007183442) - https://wokwi.com/projects/330503863007183442
> 
> ```json
> "display": {
>     "width": 128,
>     "height": 50
> }
> ```
> 
> ![image](https://private-user-images.githubusercontent.com/43933482/372631226-0a28c2d5-2e50-4f0b-a460-deebf96aef60.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5NjM5NzcsIm5iZiI6MTcyODk2MzY3NywicGF0aCI6Ii80MzkzMzQ4Mi8zNzI2MzEyMjYtMGEyOGMyZDUtMmU1MC00ZjBiLWE0NjAtZGVlYmY5NmFlZjYwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDE1VDAzNDExN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBkMDllZWZiOGRjOGVjMWNlMTljZDc3ZGY4MDI0OGQyNzgxZjQ4MmYyZWJlMGVkZGU2OTQ1YjVjMTM5OWRhMDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.dOlyzAz7LcKJIJjnL5UKL1r_60Bj1u7DgOAA-5brfa0)
> 
> > > Translated with DeepL.com

Thanks a lot. I'll try that out.