wokwi/wokwi-features #707: wokwi-slide-potentiometer doesn't respect "value" on startup.

drf5n, 2023-12-28:
**Describe the bug**
wokwi-slide-potentiometer doesn't respect "value" on startup. 

**To Reproduce**
1) Set a the value attribute per

https://docs.wokwi.com/parts/wokwi-slide-potentiometer

`      "attrs": { "travelLength": "50", "value": "512" }`

2) save the simulation
3) reload the simulation

See https://wokwi.com/projects/385381410929180673 for an example:

[image]

**Expected behavior**
I expected the pots to move to the "value" position on initial load, as they do when you edit the diagram.json file:

[image]

**Environment (please complete the following information):**
 - OS: Mac
 - Browser Chrome
 - Version 120.0.6099.109 (Official Build) (arm64)

**Additional context**
None.

drf5n, 2023-12-28:
Here's a simpler example:

1) load example from https://docs.wokwi.com/parts/wokwi-slide-potentiometer

> [Slide potentiometer + Servo](https://wokwi.com/projects/297604176384360973) - Control a [servo](https://docs.wokwi.com/parts/wokwi-servo) with a slide potentiometer

2) add a value attribute:

```
    {
      "type": "wokwi-slide-potentiometer",
      "id": "potentiometer",
      "top": 79.25,
      "left": 397.42,
      "rotate": 0,
      "hide": false,
      "attrs": {"value":"512"}
    },
```

3) Save copy, such as:

https://wokwi.com/projects/385383386673965057

4) reload copy 

5) Observe that the pot is rendered at "value":"0"

urish, 2024-01-03:
Thanks for reporting! Issue reproduced.

urish, 2024-01-07:
Fixed

closed 2024-01-07 by urish

drf5n, 2024-01-07:
Thanks, it's fixed on Chrome on Mac, but not on Safari on iphone:

https://wokwi.com/projects/385383386673965057 

gives:

[image]

urish, 2024-01-07:
Right, should also be fixed now