wokwi/wokwi-features #712: "wokwi-relay-module" Atrribute "transistor" not working

maker999, 2024-01-03:
Hi,

Relay module's attribute not working as in Docs.

According to docs "wokwi-relay-module" is supposed to have a "transistor" attribute where it determines if it's going to be active-low or active-high (default) but it doesn't change anything in the simulation.

example
`{ "type": "wokwi-relay-module", "id": "relay1", "top": -86.2, "left": 211.2, "attrs": {"transistor":"pnp"} },`

Environment:
 - OS: Windows
 - Browser: firefox
 - Version [121]

urish, 2024-01-03:
Hello, as far as I can tell it is working. 

* [Without pnp](https://wokwi.com/projects/347308007359513172):

[image]

* Same project as above, after changing "transistor" to "pnp":

[image]

maker999, 2024-01-04:
thanks. could you share how you changed the attribute?
Best Regards,
Saeed Helali

urish, 2024-01-05:
```json
    {
      "type": "wokwi-relay-module",
      "id": "tiny1",
      "top": -76.6,
      "left": -134.4,
      "attrs": { "transistor": "pnp" }
    },
```

and here's the complete project: 

https://wokwi.com/projects/386069552818017281