wokwi/wokwi-features #713: ESP32C3 UART Receive Functionality (Rust Embassy)

apollolabsdev, 2024-01-06:
In a Rust no-std embassy ESP32C3 example, the UART receiver does not process input read from the terminal to the device UART Rx pin although the input is verified. The input is verified using a logic analyzer monitoring the lines. The analyzer output shows that the expected sequence is being sent to the Rx pin.

This is a Rust embassy example in the following project:
https://wokwi.com/projects/386025413115213825

The example is expected to process the input and print to the monitor the number of bytes that were read.

 - OS: macOS
 - Browser: Both Chrome and Safari

**Some Additional Context:**
- Another thing attempted is to create a loopback on the pins and hard code a transmission sequence. This is to omit the manual input from the monitor. The same behavior remains.
- Testing the same code on different ESP32C3 hardware devkits there are mixed results.

urish, 2024-01-06:
Thanks for reporting! From local testing, the same issue happens on ESP32-C3-DevKitC-02, so it does not seem like a simulator bug. I tried typing various stuff into the terminal, including Ctrl+D, but no output:

[image]

SergioGasquez, 2024-01-08:
Update: I've modified the example: https://wokwi.com/projects/386344438491467777 and now, the generated elf works fine on real HW

urish, 2024-01-08:
@SergioGasquez Interestingly, it also works in the simulator, but once you input exactly 64 characters followed by a new line: 

e.g. inputting "SWXLHHBJRDKUWXMAFCIJUFVKMNJWMOIYSWXLHHBJRDKUWXMAFCIJUFVKMNJWMOIY" into the text box results in:

[image]

Interestingly, the real hardware works very similarly, but with a subtle difference

[image]

note how it says 64 bytes instead of 65 bytes.

Any idea where this 64 / 65 bytes number is coming from? Probably some buffer size, but which?

urish, 2024-01-08:
As for detecting the end of line condition, I looked at the code and it seems like the issue is that the simulator does not implement the AT CMD functionality that the code uses:

```rust
    uart0.set_at_cmd(AtCmdConfig::new(None, None, None, AT_CMD, None));
```

urish, 2024-01-08:
AT_CMD fix is up - [@SergioGasquez's project](https://wokwi.com/projects/386344438491467777) should now correctly detect the newlines on Wokwi

closed 2024-01-08 by urish