wokwi/wokwi-features #715: Wokwi simulator with Vs^code

leppto, 2024-01-13:
**Describe the bug**
A clear and concise description of what the bug is. 

**To Reproduce**
Steps to reproduce the behavior, including link to a simulation project where we can see the problem.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: [e.g. iOS/Android/Windows]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

leppto, 2024-01-13:
Mac os 14.2.1
VS Code 1.85.1
Wokwi Simulator 2.4.0

When running code ( https://wokwi.com/projects/386174991780008961 ) with browser, everything works, but running in VS Code, terminal "printing" is not visible and ex. servo doesn't move.

urish, 2024-01-13:
Hello, can you please share the complete project directory in a github repo and post the link here?

Usually, this is because of incorrect configuration or not compiling the project correctly

leppto, 2024-01-14:
https://github.com/leppto/RemoteGearControl

urish, 2024-01-14:
Thanks! The problem happens because the board name in platformio.ini is incorrect. Please change it as follows:

```
platform = atmelavr
board = nanoatmega328
```

Then it should work:

[image]

closed 2024-01-14 by urish