wokwi/wokwi-features #716: Download individual file

hapiel, 2024-01-15:
**Is your feature request related to a problem? Please describe.**
When working with Micropython, and I want to upload code to my physical board, it's a bit a hassle to save the python file. I can either download the project zip and extract the .py from there, or I can copy and paste the code to another editor.

**Describe the solution you'd like**
I'd enjoy a way to download an individual file.
Could be implemented in many ways, for example:

1. Hidden in the `f1` menu
2. In the dropdown menu that also allows to upload files
3. In the dropdown next to save, add 'download file'

2 or 3 are my preferred solutions.

Perhaps this is of use for those working with .ino files too.

Thanks!

urish, 2024-01-29:
Thanks for the feedback Daniel! Have you considered using Wokwi for VS Code with MicroPython? 

There's an example repo here: https://github.com/wokwi/wokwi-vscode-micropython

It might be easier this way, as you can use the same method to program both Wokwi and the physical board.

hapiel, 2024-01-29:
Thanks for your reply. That would work for many usecases I suppose, but not for the one I had in mind.

I was thinking of using wokwi for workshops, as (among many other good reasons, thank you so much for this great platform) it saves me from having to install an IDE on the participants laptop. Originally I was thinking of using the UF2 binary which can be downloaded, and upload that on a pico which can emulate a usb drive. However, I imagine Python might be even more approachable for my workshop participants.

So for my own projects, VS code would work fine, but then again for my own projects I don't really use Wokwi 😅

urish, 2024-03-20:
A workaround for workshop might be to use a service such as gitpod, where you can run Wokwi for VS Code without having to install anything on your PC.

closed 2024-03-20 by urish