wokwi/wokwi-features #721: HTTPS / TLS requests not completing on ESP32-C3

Jesse-Cameron, 2024-01-20:
**Describe the bug**
When sending a network request to endpoints using HTTPS/TLS the simulation hangs momentarily before logging: `API Error: Connection to wss://wokwi.com/api/ws/beta closed unexpectedly: code 1006` and halting. Requests to non-HTTPS requests complete successfully.

**To Reproduce**

Example repository here: https://github.com/Jesse-Cameron/trainer.

**Expected behavior**
I would expect the request the network request to complete without hanging.

**Environment (please complete the following information):**
 - OS: linux
 - Browser: `wokwi-cli`
 - Version: 0.9.0

**Additional context**

 - chip: esp32-C3
 - discord thread: API Error: Connection to wss://wokwi.com/api/ws/beta closed unexpectedly: code 1006

urish, 2024-01-21:
Issue reproduced, and there's a partial fix (the simulation no longer crashes)

urish, 2024-01-24:
Further investigation shows that it hangs in the RSA peripheral

urish, 2024-01-24:
As a workaround, you can add the following line to `sdkconfig.defaults`:

```
# CONFIG_MBEDTLS_HARDWARE_MPI is not set
```

It should disable the RSA hardware acceleration. Can you please try this and report back?

urish, 2024-01-26:
Update: RSA hardware acceleration now works correctly on ESP32-C3. Your original code should work in simulation as is:

```
Wokwi CLI v0.9.0 (abc2bb2a3530)
Connected to Wokwi Simulation API 1.0.0-20240126-gf8da8cd8
Starting simulation...
[github:SergioGasquez/wokwi-icm42670p@0.1] Initializing ICM42670P
[github:SergioGasquez/wokwi-icm42670p@0.1] Chip initialized!
[github:SergioGasquez/wokwi-shtc3@0.1] Initializing SHTC3
[github:SergioGasquez/wokwi-shtc3@0.1] Chip initialized!
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x420
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2370
entry 0x403ce000
I (32) cpu_start: Unicore app
I (32) cpu_start: Pro cpu up.
I (16) cpu_start: Pro cpu start user code
I (16) cpu_start: cpu freq: 160000000 Hz
I (16) cpu_start: Application information:
I (16) cpu_start: Project name:     libespidf
I (17) cpu_start: App version:      106da70-dirty
I (17) cpu_start: Compile time:     Jan 24 2024 23:45:57
I (17) cpu_start: ELF file SHA256:  0000000000000000...
I (18) cpu_start: ESP-IDF:          v5.1.1
I (18) cpu_start: Min chip rev:     v0.3
I (18) cpu_start: Max chip rev:     v0.99
I (19) cpu_start: Chip rev:         v0.3
I (19) heap_init: Initializing. RAM available for dynamic allocation:
I (19) heap_init: At 3FC97B30 len 00044BE0 (274 KiB): DRAM
I (20) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (20) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (21) spi_flash: detected chip: gd
I (21) spi_flash: flash io: dio
W (21) rmt(legacy): legacy driver is deprecated, please migrate to `driver/rmt_tx.h` and/or `driver/rmt_rx.h`
W (22) timer_group: legacy driver is deprecated, please migrate to `driver/gptimer.h`
I (22) sleep: Configure to isolate all GPIO pins in sleep state
I (23) sleep: Enable automatic switching of GPIO sleep configuration
I (23) app_start: Starting scheduler on CPU0
I (23) main_task: Started on CPU0
I (23) main_task: Calling app_main()
I (33) trainer::wifi: Wifi password is empty
I (43) pp: pp rom version: 9387209
I (43) net80211: net80211 rom version: 9387209
I (63) wifi:wifi driver task: 3fca41cc, prio:23, stack:6656, core=0
I (63) wifi:wifi firmware version: ce9244d
I (63) wifi:wifi certification version: v7.0
I (63) wifi:config NVS flash: enabled
I (73) wifi:config nano formating: disabled
I (73) wifi:Init data frame dynamic rx buffer num: 32
I (73) wifi:Init management frame dynamic rx buffer num: 32
I (83) wifi:Init management short buffer num: 32
I (83) wifi:Init dynamic tx buffer num: 32
I (93) wifi:Init static tx FG buffer num: 2
I (93) wifi:Init static rx buffer size: 1600
I (93) wifi:Init static rx buffer num: 10
I (103) wifi:Init dynamic rx buffer num: 32
I (103) wifi_init: rx ba win: 6
I (113) wifi_init: tcpip mbox: 32
I (113) wifi_init: udp mbox: 6
I (113) wifi_init: tcp mbox: 6
I (123) wifi_init: tcp tx win: 5744
I (123) wifi_init: tcp rx win: 5744
I (133) wifi_init: tcp mss: 1400
I (133) wifi_init: WiFi IRAM OP enabled
I (133) wifi_init: WiFi RX IRAM OP enabled
[... 29 lines cut ...]
I (10663) trainer: Hello, world!
I (10663) trainer::http: https client created
I (11293) esp-x509-crt-bundle: Certificate validated
I (14623) trainer::http: request created
I (14723) trainer::http: request submitted
I (14723) trainer::http: status: 200
I (16403) trainer::http:     <filter id="black-to-c33425">
        <feColorMatrix type="matrix" values="0.175 0 0 0.75 0  0.13 0.81 0 0.06 0  0.3 -0.31 1 0.01  0 0 0 0 1 0"></feColorMatrix>
    </filter>
</svg>
  <script type="text/javascript" src="https://www.espressif.com/sites/all/modules/syntaxhighlighter/syntaxhighlighter.min.js?s7pbz5"></script>
<script type="text/javascript" src="https://www.espressif.com/sites/all/modules/browserclass/browserclass.js?s7pbz5"></script>
  <script type="text/javascript">
    // (function (w, d, e, x) {  w[e] = function () {    w.cbk = w.cbk || [];    w.cbk.push(arguments);  };  x = d.createElement('script');  x.async = true;  x.id = 'zhichiScript';  x.src = 'https://espressif.sobot.com/chat/frame/v2/entrance.js?sysnum=8be9a70d897b4f8a890eceb29f8f77bf';  d.body.appendChild(x);})(window, document, 'zc');
  </script>
</body>
</html>
<filter id="black-to-00595a">
        <feColorMatrix type="matrix" values="1 0 0 0 0 -0.1 0.985 0 0.1 0 -0.1 -0.015 1 0.101 0 0 0 0 1 0"></feColorMatrix>
    </filter>

I (18743) trainer: Hello, world!
I (18753) trainer::http: https client created
I (19353) esp-x509-crt-bundle: Certificate validated
```

closed 2024-01-26 by urish

Jesse-Cameron, 2024-01-28:
Just confirmed that this works now. Thanks @urish !