wokwi/wokwi-features #724: Precesion settings with up-down arrow on existing gauge settings on simulation proccess.

amp-coder, 2024-01-24:
I'm always frustrated when trying to adjust the precise distance into ultrasonic sensor.
It would be nice for all of us to do some precesion settings with up-down arrow on existing gauge settings on simulation proccess.
With the existing gauge bar, it's almost impossible to do such precision adjustment. Ex. adjust the distance into ultrasonic sensor.
It would be very usefull if there are an up and down arrow to do that settings with ease like the photo.
[image]

urish, 2024-01-24:
Thanks for the feedback! 

Have you tried clicking on the slider and using the keyboard up/down keys to change the value?

amp-coder, 2024-01-24:
No but i can see that your solution is works well. Thank you Uri.

closed 2024-01-24 by urish