wokwi/wokwi-features #725: Cannot go to "your subscriptions" for my profile

geekgarage, 2024-01-29:
**Describe the bug**
Cannot go to "your subscriptions" for my profile

**To Reproduce**
Click profile img on wokwi website and click "your subscriptions"

**Expected behavior**
No clue other than it should not give me an error

**Environment (please complete the following information):**
 - OS: Windows
 - Browser: Firefox
 - Version: 122.0 (64-bit)

**Additional context**

[image]
[image]

Here is a test project so you can locate my profile
https://wokwi.com/projects/388273577509997569

geekgarage, 2024-01-29:
I can confirm it i can go into "Your Subscriptions". But i don't know If i should be able to restart my subscription.
[image]

Currently i cannot "reverse" my cancellation, and it would be nice to have the option to do so.

urish, 2024-01-29:
Thanks for the feedback! I'm not sure if our payment provider (Paddle) supports this, I'll have to check. It's possible to create a new subscription (on https://wokwi.com/club), but I do understand the value of undoing your cancellation

geekgarage, 2024-01-29:
Problem is that i bought for a year and then cancelled right after to avoid getting renewed in case it didn't provide me with value, but it actually provides me great value, so i would like to keep going whenever it runs out. But for now I'll just have to keep an eye out for when it stops saying "Club Member" in my profile and then renew, because if I do it now it want full payment for a year, today, and i know I at least have about 6 months left.