wokwi/wokwi-features #726: Custom board isn't fully saved in project

geekgarage, 2024-01-29:
**Describe the bug**
When i got to load my custom board into a project. then save it, after I've been doing wiring up components. If i then close the site and come back to it later I will have to load the custom board again before it is visible

**To Reproduce**
1. Create a project
2. Click inside the code editor, press "F1" and choose "Load custom board file..."
3. Select the directory that contains the board definition files (board.json and board.svg)
4. Board is now loaded
[image]
5. Save project
6. Close website
7. Open site and go to "My Projects" in the profile menu
8. Open the same project and have a look
[image]
9. Click inside the code editor, press "F1" and choose "Load custom board file..."
10. Select the directory that contains the board definition files (board.json and board.svg)
11. Board is now loaded

**Expected behavior**
I expect when i save the project with my custom board, that the board is saved to the project as well. 
I don't expect that when i open the project on my PC or tablet that i would have to load the custom board again every time i open it

**Environment (please complete the following information):**
 - OS: Android/Windows
 - Browser Both Chrome and Firefox
 - Version Latest for both windows and android

**Additional context**
-

urish, 2024-01-29:
Thanks for the detailed report. This is by design: the purpose of `wokwi-custom-board` is just to serve as a quick way of testing custom boards while developing them. They are not meant to be sharable (yet).

 The JSON format for custom boards will probably change at some point, and I hope to implement a UI for defining the pins, connection and board features (#302 tracks this). Until this happens, the way to go with custom boards is using the custom-boards repo that you've already found :)

closed 2024-01-29 by urish

geekgarage, 2024-01-30:
I see that you note in #302 that "Right now, we already have most of this implemented in the [wokwi-boards](https://github.com/wokwi/wokwi-boards/) project, but it's not very documented, and I perhaps not user friendly enough, so we don't see any user contributed custom boards."

Now I have contributed with a board and would love to create more. But if it takes forever to actually get them added, I can't really spend much time on it. I would love to help out and contribute to the project, but again, only if I actually see it's been taken serious and actually gets added within a reasonable time frame, not months after a PR have been made. I honestly think it's well enough documented to do custom boards as it is, also why i did https://github.com/wokwi/wokwi-boards/pull/28 to get the board I'm using in the list of available boards.