wokwi/wokwi-features #731: ESP32 ledc produces same frequency if channel 0 and 1 aere used

wvsbsp, 2024-02-07:
**Describe the bug**
I use two ledc channels to generate pwm with different PWM-frequncies. Both channels have the same PWM frequencies if channel 0 and 1 are used. Two different (correct) frequencies are generated, when i use channel 0 togehter with channel 2, 3, .... etc.

**To Reproduce**
In this example, both frequencies are different (and no multiples). Change the second #define to use channel 1 instead of 2, and both frequencies are identical (50Hz). Then swap lines 9 and 10 and both channels generate 123Hz (lastest setting "wins")
https://wokwi.com/projects/388882778029965313

**Expected behavior**
Each LEDC PWM channel should be independent of his neighbors.

**Environment (please complete the following information):**
 - OS: Win11
 - Browser Chrome
 - Version 120

urish, 2024-02-08:
Thanks for reporting! Have you tested this in hardware too?

wvsbsp, 2024-02-08:
I used different PWM frequencies in other projects with real hardware befor. I didn't notice any malfunction of the servo (that was the part that made me wonder...). But it could be that in my hardware-projects is set the servo-pwm last.
I'll check tomorrow several combinations of channels and frequencies and hook up my 4chan scope.

urish, 2024-02-09:
Thanks for checking!