wokwi/wokwi-features #733: WOKWI stops running when another browser tab is selected and used

systembolaget, 2024-02-11:
Maybe this is normal behaviour? Or is it some kind of bug? Safari and Firefox macOS.

urish, 2024-02-13:
Hi! Yeah, this is the normal behaviour - the simulation pauses when the browser tab is not visible. The simulation is resource intensive, so we don't want to end up in a situation where users leave them running in a background tab, forget about them, and then their laptop's battery drains fast because or their computer becomes slow when using other programs.

Do you have a use case where you need to keep the simulation running while the tab is not visible? Note that the browser tab doesn't have to be active - just visible (or partially visible) on the screen.

systembolaget, 2024-02-13:
Thanks, no, I only wanted to know if this is normal behaviour. All's good.

closed 2024-02-13 by urish