wokwi/wokwi-features #735: SD card fails to init on esp-idf

petermm, 2024-02-15:
**Describe the bug**
SD card fails to init on default esp32 esp-idf boards - using 100% copy of example code from https://github.com/espressif/esp-idf/blob/master/examples/storage/sd_card/sdspi/main/sd_card_example_main.c (pins/spi host changed to board values)

It does however work if one removes the `{ "builder": "esp-idf" }`  in diagram.json, suggesting SDK issue.

**To Reproduce**
https://wokwi.com/projects/389720358551651329 - use esp-idf example code provided.
Try and remove `{ "builder": "esp-idf" }` and see it working - same behaviour on wokwi-esp32-devkit-v1 board.

**Expected behavior**
Working SD card.

**Environment (please complete the following information):**
esp32 with esp-idf

urish, 2024-02-15:
Thanks for reporting!

The major difference between the Arduino builder (the one you get without a builder) is that the ESP-IDF version is v4.4.6, where's the "esp-idf" build is based on a much newer version, v5.1.2. 

I'll start investigating why the behavior is different.

petermm, 2024-02-15:
yeah, using wokwi-ci I was able to bisect:

```
      matrix:
        esp-idf-target: ["esp32"]
        idf-version:
        - 'v4.4.6'
        - 'v5.0.5'
        - 'v5.1.2'
        - 'v5.2-rc1'
        - 'release-v5.1'
```

and only v4.4.6 succeeds, all else gives the `sdmmc_sd: sdmmc_init_sd_ssr: sdmmc_send_cmd returned 0x107` error.

Maybe you are better suited than me to report upstream - feel free to close this issue at your leisure, as error is upstream.

urish, 2024-02-15:
Thanks! Well, before we mark this as an upstream issue, we have to make sure first it also happens on physical hardware.

urish, 2024-02-15:
Note: looking at the logic analyzer, I can see there's some data going back-and-forth between the the simulated ESP32 and the virtual SD Card (IDF version v5.1.2):

[image]

urish, 2024-02-15:
Also, 0x107 is probably

```cpp
#define ESP_ERR_TIMEOUT             0x107   /*!< Operation timed out */
```

urish, 2024-02-15:
I looked at the difference between 4.4.6 and 5.1.2 signal traces, and I have a suspect: seems like 5.1.2 is sending an ACMD13 command, while 4.4.6 doesn't:

[image]

I'll check our SD card implementation to see if it handles ACMD13 correctly.

petermm, 2024-02-15:
seems like multiple issues upstream:

most promising: https://github.com/espressif/esp-idf/issues/12968#issuecomment-1888668680

potential retry logic issue: https://github.com/espressif/esp-idf/pull/11297

some are even abandoning hw and moving to sd mmc: https://github.com/espressif/esp-idf/issues/12379#issuecomment-1864089045

urish, 2024-02-15:
Update: seems like our implementation is missing ACMD13. Looking at [esp-idf source code](https://github.com/espressif/esp-idf/blob/5b8d904a9b36d607aa82b95e43c571516dae4f5f/components/sdmmc/include/sd_protocol_defs.h#L412), this command should reply with a a 64 byte status buffer. Now we just need to figure out the correct contents :)

urish, 2024-02-15:
Found some documentation here: https://problemkaputt.de/gbatek-dsi-sd-mmc-protocol-ssr-register-512bit-sd-status-register.htm

petermm, 2024-02-15:
and https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/sdmmc/sdmmc_sd.c#L422

petermm, 2024-02-15:
https://github.com/espressif/esp-idf/blob/4d90eedb6ef32ff949c6f298a3d845f9f07bac92/components/sdmmc/include/sd_protocol_defs.h#L374

urish, 2024-02-15:
Thanks! Do you happen to have a physical SD card to test with? if so, can you please add the following prints after calling `esp_vfs_fat_sdspi_mount()` so we can see the values from an actual card?

```cpp
printf("ssr.alloc_unit_kb: %d\n", card->ssr.alloc_unit_kb);
printf("ssr.erase_size_au: %d\n", card->ssr.erase_size_au);
printf("ssr.cur_bus_width: %d\n", card->ssr.cur_bus_width);
printf("ssr.discard_support: %d\n", card->ssr.discard_support);
printf("ssr.fule_support: %d\n", card->ssr.fule_support);
printf("ssr.erase_timeout: %d\n", card->ssr.erase_timeout);
printf("ssr.erase_offset: %d\n", card->ssr.erase_offset);
```

urish, 2024-02-15:
No worries - I created a naive implementation that just returns a zeroed-out status buffer, and it seems to do the trick. 

Can you please test if it does the trick for you?

offtopic - I'm curious, what was the motivation behind porting the qemu tests to Wokwi? What's was missing from qemu?

closed 2024-02-15 by urish

petermm, 2024-02-15:
> Can you please test if it does the trick for you?

It mostly works! meaning unlink doesn't work:
```
[0;31mE (629) sdmmc_cmd: sdmmc_erase_sectors: sdmmc_send_cmd (ERASE_GROUP_START) returned 0x107[0m
[0;31mE (629) diskio_sdmmc: sdmmc_erase_sectors failed (263)[0m
```

I've added an unlink to https://wokwi.com/projects/389720358551651329  - so can be replicated there..

> offtopic - I'm curious, what was the motivation behind porting the qemu tests to Wokwi? What's was missing from qemu?

Just exploring and POC'ing the CI - wokwi does allow for testing _release_ packaging and artifacts, eg. is the c6 artifact correct and working - while qemu kinda have emulated special compilation things going on. 

Also experimenting with DX using VS Code sim:
Eg. adding some code for the purple button and autoreloading here: (it's erlang: atomvm.net)
it's all for fun/hobby.
https://github.com/wokwi/wokwi-features/assets/1033636/c8716fb6-8a69-45f9-802b-93d6725b8b43

petermm, 2024-02-15:
> zeroed-out status buffer, and it seems to do the trick.

maybe the ssr.erase_timeout is 0 ? and yields unlink error?

urish, 2024-02-15:
Thanks! Can you please open a separate issue for the unlink?