wokwi/wokwi-features #736: SD card esp-idf unlink fails/timeout

petermm, 2024-02-15:
**Describe the bug**
With esp-idf SD card fixed in https://github.com/wokwi/wokwi-features/issues/735 - it fails on unlink:

```
    sdmmc_cmd: sdmmc_erase_sectors: sdmmc_send_cmd (ERASE_GROUP_START) returned 0x107
    diskio_sdmmc: sdmmc_erase_sectors failed (263)
```

**To Reproduce**
Run code example here https://wokwi.com/projects/389720358551651329 

**Expected behavior**
unlink works.

**Environment (please complete the following information):**
esp32 with esp-idf 5+

**Additional context**
could be due to ssr.erase_timeout etc. not being set to proper values.

petermm, 2024-02-15:
couldn't understand why CI was green, even with the unlink errors..

from what I can see the file is deleted and unlink returns ESP_OK - but those error messages are shown.

Updated https://wokwi.com/projects/389720358551651329 with code showing it.

urish, 2024-02-15:
Alright, can you please test again?

Our SD Card implementation was missing the erase command. I guess the file could be considered "deleted" even if you don't actually erase the block that holds the data, and hence the unlink command didn't fail. 

Anyway, I just pushed a fix :-)

petermm, 2024-02-15:
tested on wokwi ci:
        - 'v4.4.6'
        - 'v5.0.5'
        - 'v5.1.2'
        - 'v5.2-rc1'
        - 'release-v5.1'

they all work, no warnings.

Great work! Th

urish, 2024-02-16:
Awesome, thanks for checking!

closed 2024-02-16 by urish