wokwi/wokwi-features #737: The boards wont align in the ceter of teh simulator window

matititam, 2024-02-21:
[image]
[image]
[image]

A refresh of the page later brings the board to the center. The links were opened from the home page.

urish, 2024-02-22:
Thanks for reporting, this doesn't reproduce for me. 

Does it still happen? Which web browser? Can you open the devtools console and check if there are any errors there?

matititam, 2024-02-22:
To recreate, 

I followed below steps. 

1. Open Wokwi home page and scroll down to ESP32 -IDF section - https://wokwi.com/esp32
2. Use the mouse scroll whell to click on the avaiable options  (ther are 6 board examples, currently). 

I am using chrome browser (Version 121.0.6167.185 (Official Build) (64-bit))
[image]
[image]

urish, 2024-02-22:
Thanks, nothing suspicious on the dev console. Does it center if you press fit?

[image]

Also, can you try checking in incognito mode? It might be some Chrome extension that is causing this

matititam, 2024-02-22:
When the part is not loaded as aligne, the options wont work No errors are created on the console while clicking on the options. The zoomin option doesn't work. No errors during this operation on the console. 
[image]

urish, 2024-02-22:
Hmm, then I'm assuming that in this case you are also not able to zoom in or drag with the mouse?

matititam, 2024-02-22:
Yp

> Thanks, nothing suspicious on the dev console. Does it center if you press fit? [...]

It doesn't happen in the incognito mode. It might a plugin then.

matititam, 2024-02-22:
> Hmm, then I'm assuming that in this case you are also not able to zoom in or drag with the mouse?

Mouse scrollbar can be used to zoomin zoomout. as well as drag. Even after dragging to the center, the zoom in zoom out or fit options doesn't work from the menu But the mouse scroll works.

urish, 2024-02-22:
> It doesn't happen in the incognito mode. It might a plugin then.

Can you try to temporarily disable some chrome extensions and see if this helps?

Also, can you try signing out of Wokwi (in normal Chrome) and see if this makes any difference?

matititam, 2024-02-22:
I had only two addons. Adblocker and a dark mode plugin. After removing darkmode plugin, i am not able to reproduce it.

matititam, 2024-02-22:
reinstalled the addon and the issue appears immediately. 
This is the plugin I used.

https://chromewebstore.google.com/detail/dark-mode-dark-reader-for/pjbgfifennfhnbkhoidkdchbflppjncb?hl=en-US&utm_source=ext_sidebar

urish, 2024-02-22:
So dark mode plugin it is. We should probably add this to the docs.

closed 2024-04-26 by urish