wokwi/wokwi-features #739: Inverter Chip example shows Missing Chip in VS Code Diagram viewer

whatisbyandby, 2024-02-23:
**Describe the bug**
When starting with the [Inverter Chip custom](https://github.com/wokwi/inverter-chip) chip example, the diagram editor shows the custom chip as "Missing Chip chip-inverter"

**To Reproduce**
Clone to project [https://github.com/wokwi/inverter-chip](https://github.com/wokwi/inverter-chip)

Open the project in VS Code with the Dev Container

Run the command "make" in the project root
[image]

Open the the diagram editor
[image]

The Chip shows as missing

When running the simulation, everything works as expected
[image]

**Expected behavior**
I expect the diagram to show the inverter chip, and the connection pins

**Actual behavior**
The chip shows as missing in the diagram editor

**Environment (please complete the following information):**

- Version: 1.86.2 (user setup)
- Commit: 903b1e9d8990623e3d7da1df3d33db3e42d80eda
- Date: 2024-02-13T19:40:56.878Z
- Electron: 27.2.3
- ElectronBuildId: 26908389
- Chromium: 118.0.5993.159
- Node.js: 18.17.1
- V8: 11.8.172.18-electron.0
- OS: Windows_NT x64 10.0.22631

**Additional context**
I'm cloning the project on an SSH host that is running Ubuntu, then connecting to this host via VS Code

whatisbyandby, 2024-02-23:
I'd be happy to take a look at the issue if someone with more context might be able to give me some direction on where to start to look for the problem.

urish, 2024-02-23:
Thanks for the detailed report! I reproduced the bug and pushed a fix, it should shortly be available as version 2.4.1 of the plugin (you'll need to restart VS Code to load the new version)

whatisbyandby, 2024-02-23:
Whoa! That was quick, I have the update, and the fix seems to be working. Thank you!

closed 2024-02-23 by urish