wokwi/wokwi-features #744: Master I2C not working properly - STM32 Nucleo C031 - VSCode - HAL (CubeMX project)

KhalilOuali, 2024-03-03:
**Describe the bug**
When simulating the STM32 Nucleo64 C031C6 using the VSCode extension, I can't get it to communicate with I2C devices (specifically the I2C LCD 20x04).

**To Reproduce**
1. Use STM32CubeMX to generate a project for the `STM32 Nucleo64 C031C6` board.
    1. Enable I2C.
    2. Enable pullups for SDA and SCL pins.
    3. Generate code for Makefile.
2. Add and configure `wokwi.toml` for debugging in VSCode.
3. Add and configure `diagram.json`, including `LCD 20x4 (I2C)`, `Logic Analyzer (8 channels)` and appropriate wiring.
4. Add code to test the I2C. Examples:
    - Download and include an I2C LCD library, and try to use it to communicate with the LCD.
    - Try to send an I2C command directly to the LCD.
    - `HAL_I2C_IsDeviceReady(&hi2c1, 0x27 << 1, 10, 1000)` twice in a row.
5. Build project using `Make`.
6. Simulate (and debug).

Results:
- Only the first `HAL_I2C_IsDeviceReady` succeeds.
- After the first transmission, no matter the type, the I2C busy flag (`__HAL_I2C_GET_FLAG(&hi2c1, I2C_FLAG_BUSY)`) is set indefinitely.
- Per the logic analyzer, after the first data transmission (`HAL_I2C_Master_Transmit` with timeout), `SCL` is stuck `low`, and `hi2c1.ErrorCode` is `32` which corresponds to `HAL_I2C_ERROR_TIMEOUT`. `Ack`s are being received for each byte.
- Without timeout, `HAL_I2C_Master_Transmit` gets stuck at `I2C_WaitOnFlagUntilTimeout`.

[Link to repo](https://github.com/KhalilOuali/Wokwi-STM32-HAL-with-I2C) with simulation tests, including multiple libraries and options.

**Expected behavior**
- The busy flag is reset when appropriate.
- SCL is pulled back up after data communication.
- No timeout.
- I2C communication is successful.
- LCD works.

**Environment**
 - Windows 10 Pro x64 Build 19045
 - VSCode v1.87.0
 - Wokwi extension for VSCode v2.4.2
 - STM32CubeMX v6.10.0
 - Arm GNU RM Toolchain v13.2 Rel1

**Additional context**
- Have tried multiple I2C LCD libraries.
- Have tried with other I2C peripherals in Wokwi (SSDI 306 OLED display).
- Have tried disabling pullups, custom pullups, disabling analog filter, interrupt mode, changing delays, changing pins, changing speed, and more.
- Have tried https://electronics.stackexchange.com/a/281046
None of the attempts worked.

urish, 2024-03-03:
Issue reproduced

urish, 2024-03-04:
Fix is up. Can you please test again and report?

KhalilOuali, 2024-03-04:
I have tried the no-lib code and it worked, many thanks!

However, I did try with the library codes as well but neither worked. Not sure why quite yet. I'll investigate and post back if I come to any conclusions.

KhalilOuali, 2024-03-04:
I have noticed two fixes required to make the libraries work:
- Send one byte at a time, no multi-bytes transmissions.
- Don't send two bytes with the enable bit on.

Example for the `lcd_hd44780_i2c.h`:
1. I added a function which takes an array of bytes and transmits one at a time.
```C
static bool transmitData(uint8_t *data, uint8_t length) {
  for (uint8_t i = 0; i < length; ++i) {
    if (HAL_I2C_Master_Transmit(lcdParams.hi2c, lcdParams.address, &data[i], 1, 100) != HAL_OK) {
      return false;
    }
  }
  return true;
}
```
2. I rewrote the init and write functions to use the above function, and to transmit the enable bit only once
```C
bool lcdInit(I2C_HandleTypeDef *hi2c, uint8_t address, uint8_t lines, uint8_t columns) {
  uint8_t lcdData = LCD_BIT_5x8DOTS;

  lcdParams.hi2c = hi2c;
  lcdParams.address = address << 1;
  lcdParams.lines = lines;
  lcdParams.columns = columns;
  lcdParams.backlight = LCD_BIT_BACKIGHT_ON;

  lcdCommandBuffer[0] = LCD_BIT_E | (0x03 << 4);
  // lcdCommandBuffer[1] = lcdCommandBuffer[0];
  lcdCommandBuffer[1] = (0x03 << 4);

  /* First 3 steps of init cycles. They are the same. */
  for (uint8_t i = 0; i < 3; ++i) {
    // if (HAL_I2C_Master_Transmit(lcdParams.hi2c, lcdParams.address, (uint8_t *)lcdCommandBuffer, 2, 100) != HAL_OK) {
    //   return false;
    // }

    // while (HAL_I2C_GetState(lcdParams.hi2c) != HAL_I2C_STATE_READY) {
    //   HAL_Delay(1);
    // }

    if (transmitData(lcdCommandBuffer, 2) == false) {
      return false;
    }

    if (i == 2) {
      // For the last cycle delay is less then 1 ms (100us by datasheet)
      HAL_Delay(1);
    } else {
      // For first 2 cycles delay is less then 5ms (4100us by datasheet)
      HAL_Delay(5);
    }
  }

  /* Lets turn to 4-bit at least */
  lcdCommandBuffer[0] = LCD_BIT_BACKIGHT_ON | LCD_BIT_E | (LCD_MODE_4BITS << 4);
  // lcdCommandBuffer[1] = lcdCommandBuffer[0];
  lcdCommandBuffer[1] = LCD_BIT_BACKIGHT_ON | (LCD_MODE_4BITS << 4);

  // if (HAL_I2C_Master_Transmit(lcdParams.hi2c, lcdParams.address, (uint8_t *)lcdCommandBuffer, 2, 100) != HAL_OK) {
  //   return false;
  // }

  // while (HAL_I2C_GetState(lcdParams.hi2c) != HAL_I2C_STATE_READY) {
  //   HAL_Delay(1);
  // }

  if (transmitData(lcdCommandBuffer, 2) == false) {
    return false;
  }

  /* Lets set display params */
  /* First of all lets set display size */
  lcdData |= LCD_MODE_4BITS;

  if (lcdParams.lines > 1) {
    lcdData |= LCD_BIT_2LINE;
  }

  lcdWriteByte((uint8_t)0x00, &lcdData);  // TODO: Make 5x10 dots font usable for some 1-line display

  /* Now lets set display, cursor and blink all on */
  lcdDisplayOn();

  /* Set cursor moving to the right */
  lcdCursorDirToRight();

  /* Clear display and Set cursor at Home */
  lcdDisplayClear();
  lcdCursorHome();

  return true;
}
```

```C
static bool lcdWriteByte(uint8_t rsRwBits, uint8_t *data) {
  /* Higher 4 bits*/
  lcdCommandBuffer[0] = rsRwBits | LCD_BIT_E | lcdParams.backlight | (*data & 0xF0);  // Send data and set strobe
  // lcdCommandBuffer[1] = lcdCommandBuffer[0];                                          // Strobe turned on
  lcdCommandBuffer[1] = rsRwBits | lcdParams.backlight | (*data & 0xF0);              // Turning strobe off

  /* Lower 4 bits*/
  lcdCommandBuffer[2] = rsRwBits | LCD_BIT_E | lcdParams.backlight | ((*data << 4) & 0xF0);  // Send data and set strobe
  // lcdCommandBuffer[4] = lcdCommandBuffer[3];                                                 // Strobe turned on
  lcdCommandBuffer[3] = rsRwBits | lcdParams.backlight | ((*data << 4) & 0xF0);              // Turning strobe off

  // if (HAL_I2C_Master_Transmit(lcdParams.hi2c, lcdParams.address, (uint8_t *)lcdCommandBuffer, 4, 100) != HAL_OK) {
  //   return false;
  // }

  // while (HAL_I2C_GetState(lcdParams.hi2c) != HAL_I2C_STATE_READY) {
  //   HAL_Delay(1);
  // }

  if (transmitData(lcdCommandBuffer, 4) == false) {
    return false;
  }

  return true;
}
```

I am not sure whether this warrants another internal fix (there could be an issue with the I2C LCD component not parsing I2C commands correctly).

urish, 2024-03-04:
Thanks for investigating! What's the quickest way for me to reproduce the issue with multi-byte transactions?

KhalilOuali, 2024-03-04:
The quickest way would probably to try one of the libraries included in the test repo.
I'll try to update it with a minimal reproduction.

KhalilOuali, 2024-03-04:
I have just pushed the changes to the test repo. You can un/comment `#define USE_MULTI_BYTE_TRANSMISSION` under `/* USER CODE BEGIN PD */` in main.c to test either mode.

It's just an edit over the nolib code, so make sure not to include the library.

urish, 2024-03-04:
Excellent - fix is up. Can you please test now?

KhalilOuali, 2024-03-04:
Nice, multi-byte transmission works now. Thanks for the quick work!

Only thing left now is that sending the enable bit twice in a row results in a transmission fail, though tbh I don't know why the library developer chose to do it anyway. A single enable bit seems to work just fine.

KhalilOuali, 2024-03-04:
I just checked the library repo and there's an issue about the topic there.

- [Original link](https://github.com/firebull/STM32-LCD-HD44780-I2C/issues/4) in russian
- [Translated link](https://github-com.translate.goog/firebull/STM32-LCD-HD44780-I2C/issues/4?_x_tr_sl=auto&_x_tr_tl=en&_x_tr_hl=en&_x_tr_pto=wapp)

Translated explanation by @firebull:

> In order to display strobe, we send the first batch of 4 bits twice. The third byte is used to send the low-order 4-bits and close the strobe. In this situation and the I2C bus speed is 100 kbit, the strobe width is ~180 µs, the pause between strobes is ~90 µs, and the pause between paired strobes is ~530 µs. In theory, and I thought so beforehand, you don’t have to extend the strobe by two bytes, you can get by with one. But in reality it turned out that 90 µs is not enough for the strobe width, but is enough for the pause between strobes. It seems that the quartz in the display operates at a lower frequency than it should according to the datasheet.

Either way, the issues discussed here are all fixed and done.
Awesome work on your end @urish. Many thanks!

closed 2024-03-04 by KhalilOuali