wokwi/wokwi-features #746: DS1307 Day Of The Week register does not work.

Koepel, 2024-03-06:
The Day Of The Week register of the DS1307 in Wokwi returns a fixed number. It can not be written.

Test: https://wokwi.com/projects/391618588603523073

Result:
```
  Test Hours    : 1=1, 2=2, 3=3, 4=4, 5=5, 
  Test Month    : 1=1, 2=2, 3=3, 4=4, 5=5, 
  Test Year     : 0=0, 1=1, 2=2, 3=3, 4=4, 
  Test DayOfWeek: 1=5, 2=5, 3=5, 4=5, 5=5,      <- returns same number
  Test MonthDay : 1=1, 2=2, 3=3, 4=4, 5=5,
```
I write a number to a register and then read it back. It is the same number for every register, except for the Day Of The Week register.
The number seems to depend on what was written to other registers.

urish, 2024-03-20:
Thanks for reporting! Yeah, the day of the week indeed depends on the date. I'm not sure what the real hardware does when you change it, is it writable? Does changing it affect any of the other fields? Perhaps someone with a DS1307 can verify.

Koepel, 2024-03-20:
According to the datasheet:
A number of 1...7 can be written.
At midnight the number is increased (7 rolls over to 1).
There is no relation with other registers.

There is not a single serious library that uses that register.

urish, 2024-03-20:
Thanks for looking this up. The implementation should now match what the datasheet says.

DarwinWasWrong, 2024-03-20:
I tested this on proteus sim. The register is writable.
It appears to allow the use/library to assign an arbitrary number to the day of the the week. You can set eg Wednesday to be a 2. then when you get the DOW back, you have code to return the day name (?) based on that number. It supports to suit week starts on a Monday not a Sunday?

urish, 2024-03-21:
Thanks for testing! It's now writable on Wokwi as well :-)

closed 2024-03-21 by urish