wokwi/wokwi-features #747: TCP connection never closes -> timeouts Wokwi-CI.

petermm, 2024-03-06:
**Describe the bug**
When doing an "active" tcp request, wokwi ci, behaves differently from real device, qemu, and even wokwi.com.

The tcp connection is stalled, and never closes, leading to a timeout error.

Tracing shows:

**on real S3, qemu, and wokwi.com(public gateway?):**
```
socket_callback netconn=0x3fcc25c4, evt=2, len=18
socket_callback netconn=0x3fcc25c4, evt=0, len=497
socket_callback netconn=0x3fcc25c4, evt=0, len=0 <---- THIS IS MISSING ON WOKWI CI.
```

and normal socket close: "Connection from 93.184.216.34:80 closed."

**on wokwi ci (private gw?):**
```
socket_callback netconn=0x3fcc25c4, evt=2, len=18
socket_callback netconn=0x3fcc25c4, evt=0, len=497
```

and timeout after 10sec on the socket: "Connection WAS NEVER closed!"

**To Reproduce**
Code is erlang on AtomVM, but surely could be replicated in C.
code is basically this:
https://github.com/atomvm/AtomVM/blob/main/examples/erlang/esp32/tcp_client_esp32.erl
You can run attached [atomvm-wokwigw-esp32s3.img.zip](https://github.com/wokwi/wokwi-features/files/14512096/atomvm-wokwigw-esp32s3.img.zip)

unzip upload on wokwi.com, see that it works.
run on img wokwi ci(private gw?) See timeout.

**Expected behavior**
TCP packets are all forwarded. Consistent behavior.

**Additional context**
Seems to be a difference between public/private gw, maybe some tcp linger options gobbling up the tcp close, but I'm no expert.

urish, 2024-03-07:
Thanks for the detailed report! Issue reproduced for me.

urish, 2024-03-07:
Fix is up - can you please test again and report?

closed 2024-03-07 by urish