wokwi/wokwi-features #748: Pico W Arduino does not connect to Wifi Wokwi-GUEST

PPATSSAR, 2024-03-08:
**Describe the bug**
The example Arduino sketch by Urish, using a Pico W, never connects to the virtual WiFi access point called Wokwi-GUEST.

**To Reproduce**
https://wokwi.com/projects/360610135224442881

**Expected behavior**
A Wifi connection after a while.

**Environment (please complete the following information):**
 - OS: Windows 10 & 11
 - Browser: Chrome & Edge
 - Version: N/A

**Additional context**
N/A

urish, 2024-03-11:
Thanks for reporting, issue reproduced

urish, 2024-03-25:
Further investigation shows that this broke once pi-pico SDK v1.5.1 was released. 

As a workaround, I downgraded the Pi Pico Arduino core to v3.2.2, which is the latest version using pi-pico SDK v1.5.0.

urish, 2025-03-29:
Fixed the simulation model, it should now work with any version of the Pico SDK.

closed 2025-03-29 by urish