wokwi/wokwi-features #751: Simulator in VSCode gets stuck on "Loading project..."

Timmmm, 2024-03-13:
On Windows 10 I created an ESP32 Rust project like this:

```
PS D:\> cargo generate esp-rs/esp-idf-template cargo
⚠️   Favorite `esp-rs/esp-idf-template` not found in config, using it as a git repository: https://github.com/esp-rs/esp-idf-template.git
🤷   Project Name: lockfw
🔧   Destination: D:\lockfw ...
🔧   project-name: lockfw ...
🔧   Generating template ...
✔ 🤷   Which MCU to target? · esp32c6
✔ 🤷   Configure advanced template options? · true
✔ 🤷   ESP-IDF version (master = UNSTABLE) · v5.1
✔ 🤷   Enable STD support? · true
✔ 🤷   Configure project to use Dev Containers (VS Code and GitHub Codespaces)? · false
✔ 🤷   Configure project to support Wokwi simulation with Wokwi VS Code extension? · true
✔ 🤷   Add CI files for GitHub Action? · false
🔧   Moving generated files into: `D:\lockfw`...
🔧   Initializing a fresh Git repository
✨   Done! New project created D:\lockfw
```

I then opened it, built it with `cargo build`. That seemed to work. I then opened `diagram.json`, clicked "Run simulation" and it gets stuck here:

[image]

There doesn't seem to be any log output (no `Wokwi` in VSCode's `Output` view, and nothing in `Wokwi Terminal`).

I'm using the latest versions of everything. Rust 1.78.0-nightly, VSCode 1.87.2, Wokwi 2.4.3.

urish, 2024-03-13:
Thanks for reporting! Mind uploading the generated project to github and sharing the link here?

urish, 2024-03-16:
Thanks! The problem seems to be incorrect part name in diagram.json. You need to change "board-esp32-c6-devkitm-1" to "board-esp32-c6-devkitc-1":

```diff
 {
     "version": 1,
     "editor": "wokwi",
     "author": "Tim Hutt <tdhutt@gmail.com>",
     "parts": [
         {
-            "type": "board-esp32-c6-devkitm-1",
+            "type": "board-esp32-c6-devkitc-1",
             "id": "esp",
             "top": 0.59,
             "left": 0.67,
             "attrs": {
                 "flashSize": "16"
             }
         }
     ],
     "connections": [
         [
             "esp:TX",
             "$serialMonitor:RX",
             "",
             []
         ],
         [
             "esp:RX",
             "$serialMonitor:TX",
             "",
             []
         ]
     ],
     "serialMonitor": {
         "display": "terminal",
         "convertEol": true
     }
 }
```

Timmmm, 2024-03-16:
Aha, that worked, thanks!! Probably would be worth printing an error message... Anyway I'll make a PR to fix the template I used.

closed 2024-03-16 by Timmmm

urish, 2024-03-16:
Thanks! yeah, it's definitely a bug that it hangs in "Loading" state instead of showing a proper error message.

urish, 2024-03-17:
Okay, we now have a very visible error message in case the board doesn't exist:

[image]

Thanks again for opening the pull request to fix the board name it upstream