wokwi/wokwi-features #752: Import from circuito.io or others AI-assisted  circuits creators

jumpjack, 2024-03-14:
Site https://circuito.io allows fast setup of wiring thanks to AI which automatically wires all the components: quite impressive! And it also generates the software.
BUT, there is a but: you cannot export the schematic diagram.

A "joint venture" between wokwi and circuito.io would be great! (if they agree to implement diagram exporting...)

urish, 2024-03-25:
Thanks! Do you know of any other similar services?

Importing from circuito is not a focus for us.

However, users reported that LLMs (such as ChatGPT) can create and modify wokwi diagrams given an example. So I can imagine one could create a custom GPT with a handful of Wokwi examples (or even a copy of the entire Wokwi docs) and it would be able to generate diagrams + code in a similar manner to circuito.io.

closed 2024-03-25 by urish

jumpjack, 2024-03-26:
Unfortunately, despite many advertisement, I can't find any site actually able to draw circuits using AI, only chatter.

But I have a better idea: circuit.io is now for sale! A banner appears sometimes when you visit the page.

I'll make some experiments with ChatGPT and wokwi.

urish, 2024-03-26:
Are you going to start your own service for generating circuits using AI?

jumpjack, 2024-03-27:
no, I am LOOKING for such service!

Anyway I did some tests with chatgpt generating diagram.json, but it comes out with just a mess.