wokwi/wokwi-features #754: Running CI with CircuitPython v9.0.0-rc.1 works on ESP32S2/S3, causes safe mode on ESP32 plain, crashes Wokwi runner with C3

tyeth, 2024-03-18:
Running this CI test suite that @georgik setup, I've modified it to use later version of CircuitPython and to publish assets to release (supplied tag on workflow run), also adding a timeout input option for wokwi.

https://github.com/tyeth/esp32-lang-lab

It crashes reliably with the C3 as a target, reporting a web socket error. C6 and H2 are untested as yet.

### Discord context:
[image]
[image]

urish, 2024-03-18:
Thanks for reporting!

I updated the github action file to upload an artifact with the final diagram.json / firmware binary that fails:

https://github.com/tyeth/esp32-lang-lab/blob/52aed28899d845508513cbfc7d5fabecc29928b3/.github/workflows/build-python-circuitpython-hello-world.yaml#L150-L156

The goal is to be able to reproduce the failure outside of github action, with the exact same inputs to Wokwi CLI.

From what I can see, it's actually simulating the ESP32-S2? 
Check out the firmware artifact down this page: [https://github.com/tyeth/esp32-lang-lab/actions/runs/8326202674]

tyeth, 2024-03-18:
Good plan!
It actually passes on the S2, but crashes on the C3, this line cycles
through updating the $TARGET variable:
https://github.com/tyeth/esp32-lang-lab/blob/52aed28899d845508513cbfc7d5fabecc29928b3/.github/workflows/build-python-circuitpython-hello-world.yaml#L79C11-L79C74

tyeth, 2024-03-18:
Just looked at your run results. Previously I was only getting that socket error with the C3,[ the S2 passed](https://github.com/tyeth/esp32-lang-lab/actions/runs/8316827435/job/22756808133#step:5:174), esp32 [plain passed](https://github.com/tyeth/esp32-lang-lab/actions/runs/8316827435/job/22756808133#step:5:201) (but serial output shows circuitpython safemode activated so no code run hence heap: <blank>), esp32s3[ passed](https://github.com/tyeth/esp32-lang-lab/actions/runs/8316827435/job/22756808133#step:5:230), C3 failed and build aborted: https://github.com/tyeth/esp32-lang-lab/actions/runs/8316827435/job/22756808133#step:5:256 - https://github.com/tyeth/esp32-lang-lab/actions/runs/8316827435/job/22756808133#step:5:271

urish, 2024-03-18:
I updated it to produce just the esp32-c3 firmware, and now the issue reproduces for me locally.

From what I can see so far, the binary simulates slowly, probably due to a busy loop, and the socket hangs up after not receiving any data for more than a minute.  Have you had a chance to test the binaries on a real esp32-c3?

tyeth, 2024-03-18:
Not yet, only just got some c3/c6's and the c3 may be faulty

urish, 2024-03-18:
Okay, I tested on a physical device and it seems to work:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0xd0
load:0x403cc710,len:0x7c8
load:0x403ce710,len:0x2518
entry 0x403cc710
Serial console setup
Adafruit CircuitPython 9.0.0-rc.1 on 2024-03-14; ESP32-C3-DevKitM-1 with ESP32-C3N4
Board ID:espressif_esp32c3_devkitm_1_n4
UID:C7FD1AAD7D0C

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Hello world!
Flash size: 8192 bytes
Minimum free heap size: 121728 bytes
Restarting in 10 seconds...
Restarting in 9 seconds...
Restarting in 8 seconds...
Restarting in 7 seconds...
Restarting in 6 seconds...
Restarting in 5 seconds...
Restarting in 4 seconds...
Restarting in 3 seconds...
Restarting in 2 seconds...
Restarting in 1 seconds...
Restarting in 0 seconds...
```

So it's an issue with the sim

tyeth, 2024-03-18:
Ah excellent, was just about to come and say, "I've got a Super Mini C3, which I ran the Seeed XAIO C3 build on from here successfully: https://circuitpython.org/board/seeed_xiao_esp32c3/ although I appreciate that's not a true recreation as I don't have the c3 devkit"

urish, 2024-03-19:
Investigated the issue (compiling CircuitPython took the better part of the day). Seems to be related to the RMT peripheral used by the neopixel:

```
(gdb) bt fu
#0  0x4038994e in vPortEnterCritical ()
    at /home/uri/p/circuitpython/ports/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:327
        state = 1
#1  0x40387e28 in xQueueReceive (xQueue=0x3fca2a18,
    pvBuffer=pvBuffer@entry=0x3fca0f8c, xTicksToWait=<optimized out>,
    xTicksToWait@entry=0)
    at /home/uri/p/circuitpython/ports/espressif/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1493
        xEntryTimeSet = 0
        xTimeOut = {xOverflowCount = 0, xTimeOnEntering = 0}
        pxQueue = 0x3fca2a18
#2  0x4206a548 in rmt_tx_wait_all_done (channel=0x3fca2900, timeout_ms=<optimized out>) 
    at /home/uri/p/circuitpython/ports/espressif/esp-idf/components/driver/rmt/rmt_tx.c:541
        i = 0
        tx_chan = 0x3fca2900
        wait_ticks = 0
        t = 0x0
        num_trans_inflight = 1
        __x = <optimized out>
#3  0x4202ca88 in common_hal_neopixel_write ()
No symbol table info available.
Backtrace stopped: frame did not save the PC
```

urish, 2024-03-19:
Issue found + fixed. Apparently, we had a wrong size for the MEM_SIZE_CHn register in CH0CONF0_REG.

Can you please try again now?

tyeth, 2024-03-20:
Thank you @urish so much, looks good. C3 generates as expected.

Not sure whats going on with the C6 / H2, but less concerned for now, will revisit another time. (They both start the simulation run, but output nothing after stating "Starting Simulation").
[image]

This is a copy of circuitpython for the c6 (8mb flash) with debug turned on (using debug build checkbox in workflow beneath with no custom flags):
[espressif_esp32c6_devkitc_1_n8-en_US-latest-debug.zip](https://github.com/wokwi/wokwi-features/files/14668391/espressif_esp32c6_devkitc_1_n8-en_US-latest-debug.zip)
And one for the H2 
[espressif_esp32h2_devkitm_1_n4-en_US-latest-debug.zip](https://github.com/wokwi/wokwi-features/files/14668401/espressif_esp32h2_devkitm_1_n4-en_US-latest-debug.zip)
The expected output with these images would be just printing "Hello world" (the default code.py contents).

You can create custom circuitpython images using this workflow (tick debug mode and get board name from /ports/*/boards/* ):
https://github.com/adafruit/circuitpython/actions/workflows/custom-board-build.yml

georgik, 2024-03-20:
@tyeth @urish Amazing work. Thank you for taking care of this issue.

urish, 2024-03-20:
Checked the C6, looks like we're hanging here, rmt again:

https://github.com/adafruit/circuitpython/blob/9.0.0-rc.1/ports/espressif/common-hal/neopixel_write/__init__.c#L141

urish, 2024-03-20:
> Checked the C6, looks like we're hanging here, rmt again:

scratch that, I got confused by the error messages. The real issue is that this port is using the USB Serial JTAG peripheral, which is not (yet) fully supported on Wokwi.

tyeth, 2024-03-20:
aha, thanks for checking @urish - good to know what's supported. Presumably that would be the case still without a debug build, one for the back burner then :)

urish, 2024-03-21:
What's the current state then? Which chips are known to work?

tyeth, 2024-03-21:
I believe the C6 should work, there's an Adafruit feather C6 board coming
soon being actively worked on, and the lead Circuitpython developer has
most of the espressif devkits so I'd expect those to work.

Circuitpython V9 (stable) has been released yesterday so it's expected that
any boards on circuitpython.org/downloads should work, although only
guaranteed support on boards sold by Adafruit (includes esp devkits,
picow's etc) - but everyone with any board gets helped in GitHub or
discord/forums too.

Maybe now's the perfect time to stick some v9 Circuitpython firmwares in
the wokwi firmware assets repo  😇

urish, 2024-03-21:
Thanks! I was actually asking about Wokwi, which esp32 chips simulate CircuitPython on Wokwi without any known issues?

tyeth, 2024-03-21:
Ah my bad. I think the S2/S3/C3 so far.
I've got no time until the weekend to look though so going on info from old
runs (RC1).
I think the esp32 might too, but the lyrat board or whatever it's pointing
at has some recovery button detection triggered causing safe mode on boot,
I'll test that with something else instead

tyeth, 2024-03-21:
> Cool, let's wait for the weekend then :)
>
> Maybe now's the perfect time to stick some v9 Circuitpython firmwares in
> the wokwi firmware assets repo 😇
>
> Last time I checked, CircuitPython had very minor usage on Wokwi.com
> compared with MicroPython. I can check again.
>
> What is the value, in your opinion, of having CircuitPython on Wokwi.com?

I'm not best placed to answer that, I don't work on it professionally, more
in Arduino land, but happy to share my 2c...

Circuitpython is a more beginner friendly version of micropython (naming
and inter-port consistency), which greatly benefits from online
connectivity, as a lot of the project guides use weather or quotes APIs
along with neopixels and maybe an additional component or two (lack of
sensors is my main worry for no interest but I'll do some later in the
year). Not all projects need WiFi though, I'd estimate probably only
40-60%, the community seems to love synths etc so expect midi madness.
I see a lot of YouTube videos doing micropython and Circuitpython tutorials
where it would help to be helpful to have the wokwi simulator included both
for participating and demonstrating. Basically wokwi is great for teaching
and learning, but not currently usable for a recent version of
Circuitpython with WiFi.

This is the reason I mainly feel it couldn't currently have any real usage
as there is little value on just the Pico using CircuitPython v6 which is
what was there when I first looked - I'm aware 8.2.7 Pico is there (v6/7 =
outdated incompatible core and library names etc).

 Not to say it isn't appreciated having some versions available on boards,
but anyone whose heard of wokwi doesn't know it did Circuitpython (often
nor micropython) as they checked it out a couple of years ago.
They were also mostly still unaware of vscode / offline usage, totally
unaware of both debugging and the F1 command pallete (to upload firmware
but so much else), and were intrigued to hear about running Circuitpython
through wokwi via CI.

I think it makes sense to not have a category yet, or maybe ever, but it
can't gain traction without people having easy access / seeing it's
supported along with some recent firmwares to run, and ideally for both
WiFi and non WiFi boards.

closed 2024-10-14 by urish