wokwi/wokwi-features #755: Simulator for PIC18F87J60 (PIC18F97J60)

rtek1000, 2024-03-19:
Hello,

  I would like to suggest that it might be useful to simulate a PIC18F87J60 version, as the ESP32 with Internet access already exists, so it shouldn't be difficult to simulate a PIC.

  The PIC18F87J60 (and others perhaps) has a very low cell endurance (100-1k) and could be a major concern for firmware development on this IC.

[image]

Note: On the official [website](https://www.microchip.com/en-us/product/pic18f87j60) of the PIC18F87J60 the documents that appear are from the PIC18F97J60.

> Feature rich PIC18 Microcontroller with an integrated 10Mbps Ethernet communications peripheral

[image]

Reference:
Datasheet page 441
http://ww1.microchip.com/downloads/en/devicedoc/39762f.pdf

urish, 2024-03-19:
Thanks for the detailed suggestion! 

Right now, we are not seeing any commercial demand for PIC18F simulation, and unfortunately, simulating a new family of microcontrollers requires a lot of effort (other than the simulation, we also have to spend time learning about the ecosystem, tooling, write tests for the different peripherals, etc.). If we see substantial commercial demand, we may look into this further.

closed 2024-03-19 by urish

rtek1000, 2024-03-19:
There is a simulator called PICSimLab, and in my opinion it was relatively well sought after in the last update.

Maybe it's easier for them to capture the market that Wokwi already has (as it takes a lot of effort to add support for new platforms), and they are expanding, before there were only a few PICs, but now there are other versions.

It might be interesting to have a partnership between Wokwi and the developer of PICSimLab, who already has a firmer knowledge of PIC microcontroller simulation.

https://sourceforge.net/projects/picsim/

[image]

urish, 2024-03-20:
Yeah, PICSimLab seems like a good fit for simulating PIC18F