wokwi/wokwi-features #756: RTC Simulator

SailorMike, 2024-03-24:
I would like the DS1307 RTC in the sim to return an actual, if dummy value when getting the temperature byte, such as "25". I know this feature is not available in the 1307; however, it does now return "0" when queried, and my projects use DS3231 which does have this feature.
Having this available would allow testing of converting deg C to deg F.

urish, 2024-03-25:
Thanks! Can you please provide a link to an example project on Wokwi that reads the temperature and displays it?

SailorMike, 2024-03-25:
Hey Uri, thanks for the reply. Here's a link to my SeaDinger 4 project, which shows the displayed temperature.
The relevant code is lines 135 - 139. "GetTemperature()" comes from the DS3231 v.1.1.2 library by Eric Ayers that is distributed from the Arduino home page (line 415). I've  attached the relevant code snippet. The "PrintNumberString" function on line 136 comes from my custom "Big_Digits" library, line 36 in the .cpp file.
Mike

https://wokwi.com/projects/392456742532761601
[https://thumbs.wokwi.com/projects/392456742532761601/social/1711330352191.png]<https://wokwi.com/projects/392456742532761601>
Sea Dinger Mk 4 - Wokwi ESP32, STM32, Arduino Simulator<https://wokwi.com/projects/392456742532761601>
Run IoT and embedded projects in your browser: ESP32, STM32, Arduino, Pi Pico, and more. No installation required!
wokwi.com

urish, 2024-03-28:
Thanks! So in the interim, I updated the DS1307 implementation to always return 25.5 °C

In the future, if we add a DS3231, this functionality will probably be removed from the DS1307

closed 2024-03-28 by urish

SailorMike, 2024-03-28:
Uri, thanks a lot! It works well in my project. A suggestion: If you do implement the 3231, make it so the temperature can be varied, perhaps as an attribute in the JSON file. If you want to see it in action, here's the link:
https://wokwi.com/projects/392456742532761601
[https://thumbs.wokwi.com/projects/392456742532761601/social/1711663054922.png]<https://wokwi.com/projects/392456742532761601>
Sea Dinger Mk 4 - Wokwi ESP32, STM32, Arduino Simulator<https://wokwi.com/projects/392456742532761601>
Run IoT and embedded projects in your browser: ESP32, STM32, Arduino, Pi Pico, and more. No installation required!
wokwi.com

Closed #756<https://github.com/wokwi/wokwi-features/issues/756> as completed.