wokwi/wokwi-features #758: [Wokwi CI] timeout issue

kassane, 2024-03-25:
Hi @urish, 

I was testing wokwi-ci in my project (esp-idf binding).

`esp32` and `esp32c3` worked, but the others had the same timeout problem (50000), interrupting the simulation.

Is there a free access limit?

https://github.com/kassane/zig-esp-idf-sample/actions/runs/8426479100

urish, 2024-03-25:
Hi! 

The timeout means that the simulation did not output the requested text in time (50 seconds of simulation). 

The most common reason is incorrect configuration of the serial output in diagram.json. Looking at your repo, I see that all the chips (other than ESP32 / ESP32C3), are missing the required connections. This is what you have:

```json
    "connections": [ ],
```

This is what you need to be able to see serial output sent through UART:

```json
    "connections": [ [ "esp:TX", "$serialMonitor:RX", "", [] ], [ "esp:RX", "$serialMonitor:TX", "", [] ] ],
```

kassane, 2024-03-25:
Wow! Thanks.
https://github.com/kassane/zig-esp-idf-sample/actions/runs/8428318332

Tomorrow, I will rebase and merge it into the project. I already managed to find a problem with esp32-p4. (I need to see how to fix it)

```bash
# esp32-p4
Run wokwi-cli --timeout "50000" --expect-text "Hello, world from Zig!" \
Wokwi CLI v0.9.1 (cfd2078ca922)
Connected to Wokwi Simulation API 1.0.0-20240319-gc9b55905
Starting simulation...
ESP-ROM:esp32p4-20230811
Build:Aug 11 2023
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0xffffffff
invalid header: 0xffffffff
```

closed 2024-03-25 by kassane

urish, 2024-03-26:
> Tomorrow, I will rebase and merge it into the project. I already managed to find a problem with esp32-p4. (I need to see how to fix it)

Regarding the ESP32-P4: unlike the other chips, the simulator needs a complete .bin file that includes the bootloader and the partition table. This limitation will go away once the ESP32-P4 is stable (no longer in preview) in the ESP-IDF.

kassane, 2024-11-13:
Fixed in:
- https://github.com/kassane/zig-esp-idf-sample/pull/15