wokwi/wokwi-features #766: Both potentiometers: Initiation bug

Pilot65, 2024-04-06:
[bugreport potentiometer.zip](https://github.com/wokwi/wokwi-elements/files/14893119/bugreport.potentiometer.zip)

See attachement for Wokwi project.

`
void setup()
{
  Serial.begin(9600);
  Serial.println("Ready...");
  //while (!Serial); // This does not help
  //delay(500); // This DOES fix the problem, but it suggests that there is a bug somwhere
}
//---------------

/*
  The problem is that for the first few readings from the potentiometer, it
  outputs its default value (0), before outputting it set value.
  Try setting the knob to max (1023) and restart... there is a number of
  erratic value "0" readings before the correct output is presented.

  The number of erratic readings differs for each time the sketch is restarted !.

  Tried changing the default value to 1023, and now the problem is "mirrored".

  The above bug is the same for both potentiometers.

  The potentiometer should NEVER output anything other than its actual position value
*/

void loop()
{
  int RawValue = analogRead(0);
  Serial.println(RawValue);

  delay(20);
}`

urish, 2024-04-06:
Thanks for the detailed bug report + easy reproduction from @Koepel.

I pushed a fix - can you please give it another go and report if it's working correctly now?

Koepel, 2024-04-06:
My test is now running perfect, everything is as expected.

closed 2024-04-06 by urish