wokwi/wokwi-features #767: Can't find "new" libraries and seems to compile wrong when it does.

leftCoast, 2024-04-08:
When developing a new library there is no real way to test it on Wokwi because Wokwi can't reliably get it integrated. (Been over a week now).

On my system, Mac running Chrome and private build server thing. I just try to compile this : https://wokwi.com/projects/393401901404691457

It should be pulling in lilParser 2.0.1 but always pulls in some 1.x version. Causing compiler failures. Function not found etc. If I try to force it to load a newer version of the library it tells me it can't find it. Others don't seem to have this issue. At one point Urish "updated all the servers" and I was able to compile and get it running. BUT it didn't really work. It was like it compiled but didn't link right or something.  Maybe it created code for a different processor? Possibly an issue with the private build server? It ran but seemed like it would call the wrong functions. So, I went back to the work bench, setup an UNO with the LEDs and ran the above code on it. Compiled and worked fine. Coming back I can't get it to compile at all. It's back to trying to load the 1.x version.

 - OS: Mac 10.13.6
 - Browser Chrome
 - Version 116.0.5845.187 (Official Build) (x86_64)

I noticed my Breakout game is not running right anymore either. At least for me. I'd not changed anything on that for a long time and now it's locking up mid game and stuff like that. I wonder if these two things are related? https://wokwi.com/projects/320027251914572370

leftCoast, 2024-04-09:
Made a change. Reloaded it about 5 to 6 times and.. It's all working.

leftCoast, 2024-04-09:
And this morning, the issue is back.

leftCoast, 2024-04-10:
How hard would it be to disable the private build server thing on mine? I'd like to try it without.

leftCoast, 2024-04-10:
You know, that "?build-cache=disable" Thing isn't working for me. Even now if I mak a small change and try to recompile it fails. I tried the disable thing and that made no difference.

urish, 2024-04-10:
sure, let's try that - I disabled it for you

leftCoast, 2024-04-10:
I think that fixed it. If it finds that it's compiling something it compiled in the past. Identical text. I see the issue. Like it has multiple compile caches? I donno' not being a server guy I'm just guessing in the dark here.

I'll set one up from scratch and see what it does.

Thanks for looking into this.

leftCoast, 2024-04-10:
Made one from scratch. Compiled for over 4 minutes. Couldn't cancel it. Clicked the "learn more" button and when I came back it completed and was successful.

Now I"ll  need to update another library to see if it's completely gone. That'll take awhile.  I guess we can let this rest for a bit?

urish, 2024-04-12:
Sure, feel free to reopen the issue if it strikes again.

closed 2024-04-12 by urish

leftCoast, 2024-04-16:
Can I either get the the private build server turned back on, or my compile time lengthened? I'm working one of my huge experiments and can't compile it. Really, you should like it if I can get it to work.

urish, 2024-04-16:
Well, the private build server has been shut down for good. I'll see what we can do about getting the build time lengthened, it's a bit tricky. 

How about compiling it on your own computer and using Wokwi just for the simulation?

leftCoast, 2024-04-17:
That's even trickier. I don't currently have any way to compile for the pi Pico. Or at least I've never done the learing curve for it.