wokwi/wokwi-features #768: HTTP-GET request causes Wokwi for VS Code to hang

sivar2311, 2024-04-13:
**Describe the bug**
HTTP-GET request causes Wokwi for VS Code to hang

This project https://wokwi.com/projects/393020133767191553 runs fine in the browser.
But in VS-Code the simulation hangs after the http request was made.
CPU load increases and VS Code / Wokwi simulation becomes unresponsive in some parts. 
Stopping / closing the simulation does not help. Only a restart of VS Code stops high CPU load.

**To Reproduce**
`platformio.ini`
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
```

**Expected behavior**
Expected output:
```
WiFi Connected!
10.10.0.2
200
{
  "userId": 1,
  "id": 1,
  "title": "delectus aut autem",
  "completed": false
}
```

Receive output:
```
WiFi Connected!
10.13.37.2
```

**Environment (please complete the following information):**
 - OS: Windows 11
 - Browser Chrome
 - Version 123.0.6312.123

- VS Code: 1.88.1
- PlatformIO Core, Version 6.1.14
- Espressif32 Platform Version: 6.6.0
- Arduino Core Version: 2.0.14

**Additional context**
Compiled binary: [firmware.zip](https://github.com/wokwi/wokwi-features/files/14965694/firmware.zip)

urish, 2024-04-13:
Thanks for the detailed report, I partially reproduced it here.

Does "F1" -> "Restart extension host" help too?

sivar2311, 2024-04-14:
It helps to the effect that the simulation can be restarted. 
However, the CPU load remains high and increases even more. 
A look at the Task Manager shows that VS Code now has even more tasks.

Result: The situation gets worse (and the HTTP request is still not executed correctly).

sivar2311, 2024-04-23:
I am currently in a situation where I need an HTTP request.
Has there been any progress in solving the problem?

urish, 2024-04-23:
There's a potential workaround by using the Private IoT gateway externally, I'll share the details later today

urish, 2024-04-23:
So the workaround is to download and run the [Private IoT Gateway](https://docs.wokwi.com/guides/esp32-wifi#installation), then add the following to your wokwi.toml:

```
[net]
gateway="ws://localhost:9011"
```

Start the simulation - and it should successfully establish the connection. I just tested it on my own machine and it worked well:

[image]

sivar2311, 2024-04-24:
Thank you very much!
That workaround helps.

urish, 2024-05-02:
Fixed in Wokwi for VS Code v2.4.4. Thanks again for the super detailed report, it was very helpful!

closed 2024-05-02 by urish