wokwi/wokwi-features #770: Ultrasonic sensor bug

Pius171, 2024-04-20:
**Describe the bug**
When using an ultrasonic sensor, if the echo pin is set to output the sensor still works correctly although, it should be an input

**To Reproduce**
use the ultrasound sensor and set the echo pin to output

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: Windows
 - Browser :chrome
 - Version:

urish, 2024-04-23:
Thanks for reporting! Can you please provide a link to a Wokwi project that demonstrates the issue?

Pius171, 2024-05-05:
try any ultrasonic sensor example and make the echo pin output, it would still work normally

closed 2024-10-14 by urish