wokwi/wokwi-features #771: UF2 download not accepted by Pi Pico / RP2040 bootloader.

JamesNewton, 2024-04-23:
**Describe the bug**
https://docs.wokwi.com/parts/wokwi-pi-pico#exporting-uf2-binary
doesn't work. As in, the UF2 file is not accepted by the Pi Pico bootloader. 

**To Reproduce**
Project is here:
https://wokwi.com/projects/395639736344965121
chip is (now) a bog standard RP Pico 2020 from them (verified good via Circuit Python and Arduino IDE)

- Open project
- Start simulation, wait for compile, verify simulated LED does blink.
- Stop simulation
- Click in editor
- Press F1
- Scroll to "Download UF2 Binary" and click it.
- Hold down bootloader button on RP Pico and plug it in.
- Drag firmware file to the Flash drive "RPI-RP2" when it appears.
- Eject the drive
- Power cycle RP2 /without/ holding down the bootloader button

**Expected behavior**
EXPECTED: LED on RP2 blinks. FOUND: It doesn't.

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 16.04 and 22.04
 - Browser Chrome 124.0.6367.60
 - Version (of Wokwi? latest? This was done from the web page so whatever it served me) 

**Additional context**
Add any other context about the problem here.

urish, 2024-04-23:
Thanks for the detailed report! Uploaded a fix, can you please check again now?

JamesNewton, 2024-04-23:
Fixed. Thank you!