wokwi/wokwi-features #782: Has the slide potentiometer the VCC and GND pins mixed up ?

Koepel, 2024-05-09:
A slide potentiometer has a carbon strip with pins at the ends. A slider moves along the carbon strip.

In the documentation is this picture. It shows the slider at it lowest position.
[image]
The slider is connected to the lower-left pin. The upper-left is of course the GND pin and the VCC is the pin on the right.

In the Wokwi simulation it is the other way around.
[image]

Documentation: https://docs.wokwi.com/parts/wokwi-slide-potentiometer
My test in Wokwi: https://wokwi.com/projects/397414339706158081

If the VCC and GND pins are changed, then all the wiring of all the projects with a "wokwi-slide-potentiometer" will change. I think that is still better than adding an other slide potentiometer.

urish, 2024-05-09:
🤦‍♂️

The alternative would be to make the left side high, the right side low (to match the existing pinout).

urish, 2024-05-09:
Looking at the original pull request (which introduced the slide potentiometer), it seems like this issue was there from the very beginning: https://github.com/wokwi/wokwi-elements/pull/63/files#diff-5611a9ccd27b0203c136edee13127d7d11b638ebf8bf610e5df1d691acaab9ffR12-R14