wokwi/wokwi-features #788: ESP32-C3 `std` UART Logging Issue

apollolabsdev, 2024-05-20:
Using ESP32-C3 and `no-std`, attempting to log to the console using UART0 alongside `println` does not work. The code pauses without any output printed. 

This is a follow-up to the closed issue: esp-rs/esp-hal#1531 . 

For the non-Rust ESP32-C3 devkits, both UART and `esp_println` share the same lines, however, this behavior does not seem to reflect on Wokwi. I was not able to verify on hardware myself with a different devkit as I only have access to a Rust board.

urish, 2024-06-08:
Thanks for reporting! If I understand correctly, the prints go to the Serial JTAG UART?

apollolabsdev, 2024-06-08:
Correct. For the ESP32-C3s. Though the devkitM utilizes an external USB-UART bridge ([schematic](https://dl.espressif.com/dl/schematics/SCH_ESP32-C3-DEVKITM-1_V1_20200915A.pdf)).

urish, 2024-10-14:
Update: we now have support for logging to the serial JTAG UART. It can be enabled by adding an `attr` to diagram.json:

https://docs.wokwi.com/guides/esp32/#usb-cdc-serial-over-usb-support

closed 2024-10-14 by urish