wokwi/wokwi-features #789: A LCD via the I2C bus on STM32 boards stopped working in Arduino mode.

Koepel, 2024-05-20:
In the Discord channel was mentioned that the I2C bus does not work for the STM32 board in Arduino mode.

There is more than one problem:
1. The local files for the I2C LCD display with "Wire2" do no longer work.
2. There seems to be no longer a compatible I2C LCD library .

**Large STM32 Nucleo64 C031C6 board:**
Example for old situation: https://wokwi.com/projects/365421666018061313
When I do a `Serial.println(SDA)` and `Serial.println(SCL)` then the default pins are: SDA = 14 and SCL = 15.

**Small STM32 Nucleo32 L031K6:**
Example for old situation: https://wokwi.com/projects/367245102561225729
With `Serial.println(SDA)` and `Serial.println(SCL)`, then the default pins are: SDA = 4 and SCL = 5.

Tests: A sketch with a I2C Scanner on the default SDA and SCL pins work. A display and a RTC module are detected. 
However, the "LiquidCrystal I2C" library does not work for the display, and even the "hd44780" library does not work. I can not find any STM32 project with a working I2C LCD display. 
I don't know if a OLED display works. The Adafruit GFX library with the Adafruit SSD1306 does not compile.

Koepel, 2024-05-20:
Test with a DS1307 to pins 4,5 (small board) or 14,15 (large board):
```
#include <Wire.h>

void setup() 
{
  Wire.begin();

  Wire.beginTransmission(0x68);
  Wire.endTransmission();

  delayMicroseconds(50);

  Wire.beginTransmission(0x68);
  Wire.write(0);
  Wire.endTransmission();
}

void loop() {}
```

Result for both boards:
[image]

The SCL is stuck to low when a databyte should be transmitted. That explains that only a I2C Scanner works.

urish, 2024-05-26:
Issue reproduced - seems to be a regression introduced by #744