wokwi/wokwi-features #793: Simulation not starting

lucasssvaz, 2024-05-26:
**Describe the bug**
Sometimes the simulation apparently won't start. (Maybe it is taking way too long to start but it times out before that)

**To Reproduce**
No specific steps, sometimes the Wokwi CLI won't start the simulation after being called by pytest:

```
 =========================== short test summary info ============================
FAILED tests/validation/democfg/test_democfg.py::test_cfg - pexpect.exceptions.TIMEOUT: Not found "Hello cfg!"
Bytes in current buffer (color code eliminated): Wokwi CLI v0.9.1 (cfd2078ca922) Connected to Wokwi Simulation API 1.0.0-20240518-gdbbf1224
Please check the full log here: /tmp/pytest-embedded/2024-05-19_02-19-10-065724/test_cfg/dut.log
====================== 1 failed, 14 deselected in 30.18s =======================
```

**Expected behavior**
No hanging or big delay before starting the simulation.

**Environment (please complete the following information):**
 - OS: CI (ubuntu-latest)
 - Version: CLI 0.9.1

**Additional context**
Some places where this happened:

https://github.com/espressif/arduino-esp32/actions/runs/9144334087/job/25142139131
https://github.com/lucasssvaz/arduino-esp32/actions/runs/9240086259/job/25420033517?pr=12
https://github.com/lucasssvaz/arduino-esp32/actions/runs/9240024519/job/25419849482
https://github.com/lucasssvaz/arduino-esp32/actions/runs/9236780292/job/25413060095
https://github.com/lucasssvaz/arduino-esp32/actions/runs/9236847049/job/25413184000

urish, 2024-05-26:
Hi Lucas, thanks for reporting!

I've just launched a new experimental action that runs the CI Server next to the wokwi-cli, inside the GitHub runner itself (using docker).

To use it, simply add the following lines to your workflow, anywhere before using the wokwi-cli:

```yaml
      - name: Wokwi CI Server
        uses: wokwi/wokwi-ci-server-action@v1
```

Can you please test it and report if you are observing an improvement?

lucasssvaz, 2024-05-27:
This will be added to https://github.com/espressif/arduino-esp32/pull/9696
Let's see how it behaves

lucasssvaz, 2024-05-27:
Still happens but there's no API errors anymore:

https://github.com/espressif/arduino-esp32/actions/runs/9258209420/job/25467895329
https://github.com/espressif/arduino-esp32/actions/runs/9258209420/job/25467909242

urish, 2024-05-27:
Thanks for testing! Good to see the API errors disappeared.

Seems like the new issue is only happening with the WiFi test on the ESP32? 

Can you please add the following to the action yaml, right after running the actual test?

```yaml
      - name: CI Server Logs
        if: always()
        run: |
          docker logs -t $(docker ps -q)
```

This should let us see if the CI server is reporting anything suspicious when the failures happen.

lucasssvaz, 2024-05-27:
@urish Just ignore my previous message 😆  I forgot that it will only run the updates once it is merged to master.

In my fork it is working all good for now

lucasssvaz, 2024-05-28:
Idk what happened but the server docker container started failing in all tests:

[image]

https://github.com/lucasssvaz/arduino-esp32/actions/runs/9259955226/job/25472955911?pr=14

urish, 2024-05-28:
Are you running inside a GitHub hosted runner or self-hosted runner?

urish, 2024-05-28:
Nevermind the question - the issue is that you are running the wokwi-ci-server-action twice.

lucasssvaz, 2024-05-28:
Oops, my bad. I merged the changes from upstream but didn't see that duplication. Sorry about that

closed 2024-07-24 by lucasssvaz